Ultimaker/Cura #19636: Native CAD import

axel33440, 2024-09-10:
### Is your feature request related to a problem?

Yes, the issue is that the CuraInventorPlugin is not compatible with the latest version of Cura, causing slower and less reliable file imports from Inventor. This disrupts my workflow and leads to several inefficiencies.

### Describe the solution you'd like

I would like the CuraInventorPlugin to be compatible with the latest version of Cura, allowing for faster, more reliable file imports from Inventor. Ideally, it should also retain features like model updates with a simple F5 refresh, without needing to save the model again.

### Describe alternatives you've considered

I’ve considered using Cura Enterprise for importing Inventor files directly, but it’s slower and less reliable than the plugin. It also lacks features like real-time model updates with F5, which the plugin provides. Therefore, the plugin remains the best solution for my workflow.

### Affected users and/or printers

The affected users are those who use Autodesk Inventor for design and rely on Cura for slicing, particularly in workflows where frequent model updates and quick file imports are essential. This impacts anyone needing efficient and reliable integration between Inventor and Cura, regardless of the 3D printer model used.

### Additional information & file uploads

I don’t have specific files to upload, but I can provide additional details if needed. The issue primarily involves the lack of compatibility between the CuraInventorPlugin and the latest version of Cura (5.8). The key challenges are slower file imports, failed imports, and missing features like real-time updates with F5, which worked well in previous versions with the plugin. Let me know if further information is required!

GregValiant, 2024-09-11:
The plugin was developed by a contributor (Theopikar) whose Git page is [HERE](https://github.com/thopiekar).
He also had one for Solid Works and that has gone away as well.

Cura 5.0 involved massive changes to the program.  Along with the introduction of the Arachne Engine (variable line width), a second huge change was to move from QML 5.0 to QML 6.0 (I think that was done to accomodate MAC systems).
It was the QML change that caused all of the plugins to need rework.  QML is the resource that Cura uses for dialogs and setting boxes, checkboxes, etc.

axel33440, 2024-09-11:
Yes, I managed to find Theopikar's contact information and reached out to him via email, but I haven’t received a response yet. I fear he may no longer be active, as his last contributions seem to date back several years.

I understand that significant changes were made in Cura 5.0, including the optimized trajectory calculations, which noticeably reduce print time. However, I’m still facing this issue, and I’m sure others using different modeling software are encountering similar challenges.

Is there any ongoing work or alternative solutions that might help resolve this, given the current plugin limitations?

GregValiant, 2024-09-11:
I also tried contacting him without any response.  There have been Github contributions under that name this year, but not for Cura.  I think he may have moved on.

Coming up with a workable solution requires that the developer be fluent in both Cura, and whatever CAD software programming language is involved.  A "Solid Works" solution would probably not work with "Inventor".  The whole thing appears to be a lot of work for little reward.  If you have a copy of the original Inventor plugin, you might be able to find someone to alter it for use with QML 6.  That's probably the easiest way to go but open source or not - I would think a developer would not want to step on someone else's toes.