Ultimaker/Cura #19638: Slicing Failed (Cura 5.8.0)

BrianI1977, 2024-09-10:
### Cura Version

5.8.0

### Operating System

LMDE 6 "Faye" (Debian 12)

### Printer

Ender 3 S1 (Sonic Pad)

### Name abnormal settings

crashes even with default profiles.

### Describe model location

Flat on build plate 

### Describe your model

Does not slice on 5.7.0 or 5.8.0

### Add your .zip here ⬇️

[CFFFP_rc license plate front.stl.3mf.zip](https://github.com/user-attachments/files/16955243/CFFFP_rc.license.plate.front.stl.3mf.zip)

GregValiant, 2024-09-10:
Thanks for the report.
You will have to wait for someone from the Cura team to take a look.  I didn't have any trouble with my Windows installation so this might be a Linux specific problem.
[image]

BrianI1977, 2024-09-13:
As a follow up, I did a clean install of my linux distro, and did a fresh install of the Cura 5.8.0 Appimage along with reinstalling the printer (Creality Ender 3 S1, using Sonic Pad).

Cura appears to be working now.
[image]

FireflyJar, 2024-09-15:
is there a way to remove all Cura related stuff and reinstall without doing a clean linux reinstall?  broke mine somehow, affected older versions of Cura as well.

FireflyJar, 2024-09-15:
update:  reinstalled my linux mint 22, kept my /home files and now Cura works, something broke it and will try to keep track of what I do this time and have TimeShift backups to roll back to.