Ultimaker/Cura #19640: 5.8.0 Infill after change pattern doesn't work Using the standard Fast - Engeneering -0.2mm profile

marcovalstar, 2024-09-11:
### Cura Version

5.8.0

### Operating System

Windows 11

### Printer

Ulitmaker S5

### Reproduction steps

I make a new project and choose my printing profile
Than i load my models (2 models that are merged because of printing 2 different colors)
Then i sliced the model.
Not satisfied from the result i change the infill dencity and/or the pattern but that doesn't have any effect after re-slicing

### Actual results

Using the standard Fast - Engeneering -0.2mm profile
The re-slicing is the same as the result before the changes

### Expected results

I expect the infill changed from for example 25% to 75% and the standard pattern (grid) changes to the choosen (Gyroid)

### Add your .zip and screenshots here ⬇️

[Change_infill.zip](https://github.com/user-attachments/files/16959551/Change_infill.zip)

GregValiant, 2024-09-11:
Thanks for the report.
We ask for project files so we get your Cura settings, your printer, and the model.  Without it I would be guessing and I respectfully refuse to do that.

marcovalstar, 2024-09-11:
[Error_slicing_VRT.zip](https://github.com/user-attachments/files/16964112/Error_slicing_VRT.zip)

Sorry for not given all the information. Here it is.

GregValiant, 2024-09-11:
There is so little infill in the letters, and the infill is just a single layer.  It's hard to tell what is going on.
This is my slice of the project with no changes (E2 is Gyroid at 25% and E1 is Triangles at 20%)
[image]
When I change E2 to 75%:
[image]

When I scale the model up it slices like this which appears is correct.  This is with E2 left at your 25% infill density.
[image]

I don't see a bug here and I can't duplicate the problem.  Your project does not contain the settings you describe above and the CuraProfile you provided does not contain the settings you describe above.
When I change the infill density, the infill density changes.  The infill areas of the letters are so small and shallow that they would not require any infill at all.

I'm going to remove the bug from this.  I'm inclined to close it as well.  Let me know if you feel that is premature.

GregValiant, 2024-09-11:
I can't duplicate this.  I change the extruder settings and they are reflected in the slice.  I'm using your project, I just scaled it up a bit so the infill areas are larger.
I have changed the infill pattern and density a bunch of different ways and it always appears in the print per my settings.
I'll put the bug label back on this for you and someone from the Cura team ( @HellAholic ??) will take a look.  

This happens to be with "Lines" and a custom sine wave infill I designed.  Everything worked as expected.
[image]

Here I've switched patterns and density and tile size from one part to the other.
[image]

marcovalstar, 2024-09-12:
You are so right i feel kind of stupid now :-(.
That was indeed the problem. I should check the extruder first before changing the settings.
Excuses for the time you spend on me.
Good thing is i learnd something thanks.

GregValiant, 2024-09-12:
I'll go ahead and close this then.

closed 2024-09-12 by GregValiant