Ultimaker/Cura #19647: Could not probe OpenGL.  This program requires OpenGL 2.0 or higher.  Linux Cura will no longer boot after year, all versions

jotjeff, 2024-09-13:
### Cura Version

5.8.0

### Operating System

Linux Kali (Debian)

### Printer

Creality Ender 3 Max

### Reproduction steps

I have used Cura over a year now, with no problems on Kali Linux, which is a Debian distribution.
All I did was upgrade Kali to the latest version, which is 2024.3.    After upgrading, it will no longer run ANY version of Cura, I have tried loading every AppImage you have online.
My thinking is that everyone who upgrades their Linux is going to start getting this same error, as my hardware is exactly the same.   But now Cura is no longer compatible with Linux, after it is upgraded.

I have a Gateway PC.   

Here is my hardware:
└─$ lspci  
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v2/3rd Gen Core processor DRAM Controller (rev 09)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v2/3rd Gen Core processor PCI Express Root Port (rev 09)
00:02.0 VGA compatible controller: Intel Corporation Xeon E3-1200 v2/3rd Gen Core processor Graphics Controller (rev 09)
00:14.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB xHCI Host Controller (rev 04)
00:16.0 Communication controller: Intel Corporation 7 Series/C216 Chipset Family MEI Controller #1 (rev 04)
00:19.0 Ethernet controller: Intel Corporation 82579V Gigabit Network Connection (rev 04)
00:1a.0 USB controller: Intel Corporation 7 Series/C216 Chipset Family USB Enhanced Host Controller #2 (rev 04)
00:1b.0 Audio device: Intel Corporation 7 Series/C216 Chipset Family High Definition Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 7 Series/C216 Chipset Family PCI Express Root Port 1 (rev c4)
00:1c.3 PCI bridge: Intel Corporation 7 Series/C216 Chipset Family PCI Express Root Port 4 (rev c4)
00:1d.0 USB controller: Intel Corporation 7 Series/C216 Chipset Family USB Enhanced Host Controller #1 (rev 04)
00:1e.0 PCI bridge: Intel Corporation 82801 PCI Bridge (rev a4)
00:1f.0 ISA bridge: Intel Corporation B75 Express Chipset LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation 7 Series/C210 Series Chipset Family 6-port SATA Controller [AHCI mode] (rev 04)
00:1f.3 SMBus: Intel Corporation 7 Series/C216 Chipset Family SMBus Controller (rev 04)
03:00.0 SATA controller: JMicron Technology Corp. JMB363 SATA/IDE Controller (rev 03)
03:00.1 IDE interface: JMicron Technology Corp. JMB363 SATA/IDE Controller (rev 03)
--------------------------------------------

So it is not an issue of meeting minimum requirements.   It seems something is wrong.
It also does not seem to matter which version of Cura.   I have tried them all.  All of them bring up the error message,
'This program requires OpenGL 2.0 or higher'.

There appear to be many fixes on Youtube, but all of them are for Windows.  I do not see any fix for this error message for Cura running on Linux, but I imagine it must be something similar. 
One mentioned placing an OpenGL32.dll file in the executable directory.   But again that was for Windows.
I have downloaded that fix file here:

https://www.youtube.com/watch?v=1T2zTgK9NME
However, I have Linux.   I do not know where if any place, the dll file should go.

I would much appreciate help.  As I said I've been using Cura for over a year, and have lots of programs but now suddenly I can no longer use it anymore.
If there is some kind of fix, only Cura Ultimaker would know about about it.   No one else would know.   It is only your software that is not loading, everything else works fine.

Thanks much

### Actual results

All of them respond with a pop up window saying, 'Could not probe OpenGL.  This program requires OpenGL 2.0 or higher.  Please check your video card drivers'.     All versions of Cura on your website generate exactly the same error.  
When closing window, exits and nothing will load.

Some more information-
When I run my original Ultimaker Cura AppIimage version 530 from a Linux command prompt- I get this error information on my screen before the OpenGL32 error window pops up:

└─$ ./cura.AppImage530
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_static_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so
MESA-LOADER: failed to open crocus: /usr/lib/dri/crocus_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: crocus
MESA-LOADER: failed to open crocus: /usr/lib/dri/crocus_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: crocus
MESA-LOADER: failed to open zink: /usr/lib/dri/zink_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
MESA-LOADER: failed to open crocus: /usr/lib/dri/crocus_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: crocus
MESA-LOADER: failed to open crocus: /usr/lib/dri/crocus_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: crocus
MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
                   
                   
                   
                   
                   
                   
                                       
I hope this is useful in determining what is causing this bug.
Thanks again.

### Expected results

Cura was supposed to start as it always had.

### Add your .zip and screenshots here ⬇️

[image]

drew3199, 2024-09-30:
I'm experiencing the same issue on Fedora 40.  I updated from 5.8.0 to 5.8.1 and am getting the same error.  When trying to run from terminal, it output 'geqt.qpa.wayland: EGL not available'.

smartavionics, 2024-10-01:
Hi, from the errors you quote above, it looks like one problem is caused by the appimage bundling a version libglib that is incompatible with your updated system. Whether that is also causing those dri library errors, I don't know.

My Cura releases (see https://github.com/smartavionics/Cura/releases) are based on the old Cura build system and don't include libglib so they may still run on your system. Unfortunately, they are based on Cura 4.

victorem1, 2024-10-25:
I've encountered this issue too on Manjaro Linux (using KDE if it matters). Interestingly, it fails to load with Wayland, but succeeds with X11. 

Wayland startup logs:
```
qt.qpa.wayland: EGL not available
qt.accessibility.atspi: Error in contacting registry: "org.freedesktop.DBus.Error.Disconnected" "Not connected to D-Bus server"
```

X11 startup logs:
```
qt.qml.typeresolution.cycle: Cyclic dependency detected between "file:///tmp/.mount_UltiMaVa1XHB/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMaVa1XHB/share/cura/resources/qml/Actions.qml"
file:///tmp/.mount_UltiMaVa1XHB/share/cura/resources/qml/PrintSetupSelector/ProfileWarningReset.qml:16:5: QML ProfileWarningReset: Binding loop detected for property "width"
file:///tmp/.mount_UltiMaVa1XHB/share/cura/resources/qml/PrintSetupSelector/ProfileWarningReset.qml:16:5: QML ProfileWarningReset: Binding loop detected for property "width"
```
but it otherwise runs.

I tried the workaround from issue https://github.com/Ultimaker/Cura/issues/19630, setting QT_QPA_PLATFORM=xcb so that it'd run under Xwayland, but it didn't work as-is, and I don't know enough to glue the missing pieces together. 

I suppose there's some API / portal permission error with Wayland being more restrictive?

There's another workaround on that issue, running as sudo, which works for a while, until I try to import an stl to print, at which point it crashes:
```
zsh: segmentation fault  sudo ./UltiMaker-Cura-5.8.1-linux-X64.AppImage
```

Thoughts or insights appreciated. Let me know what more info would be helpful if any.

laverdone, 2024-10-31:
The problem was the libwayland embedded on AppImage, extract it delete libwayland and recompact it.

rephlex, 2026-10-25:
Kali tracks Debian testing, so the trixie builds of a signed community repo my company maintains should be your match: Cura 5.13.0 as a native .deb, built against system libraries rather than an AppImage bundle that mis-detects your GL stack:

```
curl -fsSL https://mirror.plustech.de/plustech.gpg | gpg --dearmor -o /usr/share/keyrings/plustech.gpg
echo "deb [signed-by=/usr/share/keyrings/plustech.gpg] https://mirror.plustech.de/debian/ plustech-trixie main" > /etc/apt/sources.list.d/plustech.list
apt-get update && apt-get install ultimaker-cura
```

https://mirror.plustech.de/ - community repo, not UltiMaker-official, sources published.