Ultimaker/Cura #19651: Slicing a cylinder makes rendering bug out

ignas-rss, 2024-09-13:
### Cura Version

5.8

### Operating System

Mac OS 14.6.1

### Printer

VEHO-600

### Reproduction steps

1. Add cylinder - any of these should do:
https://www.dropbox.com/scl/fo/zzjgylj1snf6bn60sadho/ADpx2Pngeqcit93kHr4K64k?rlkey=qps6scuk812bwats4mhxxpxsb&st=osccchp8&dl=0
2. Try slicing
3. Preview 
4. Weird result
[image]

### Actual results

Hard to describe that like the image shows
[image]

### Expected results

Sliced cylinder

### Add your .zip and screenshots here ⬇️

[cylinder.3mf.zip](https://github.com/user-attachments/files/16998691/cylinder.3mf.zip)
[cura.log](https://github.com/user-attachments/files/16998711/cura.log)

GregValiant, 2024-09-13:
Thanks for the report.
You have successfully re-discovered what I call the "Peter Maxx" bug.  So far it remains illusive.  I've only had it happen once and that was on my previous computer.  It won't duplicate for me with your models.
[image]

This one is my favorite:
[image]

I like this one as well:
[image]

This bug has showed up very infrequently over the last 8 or 10 versions of Cura.  It doesn't affect the gcode, but it looks really bizarre when it happens.  The Cura team will take a look.

ignas-rss, 2024-09-14:
I was initially reproducing it consistently even with app restarts, but now seems to stop happening after a bit even without restarting the app

closed 2024-09-14 by ignas-rss

GregValiant, 2024-09-15:
That's what has happened before.  The bug appears and disappears apparently at random which makes chasing it down near impossible.  It's like trying to find an intermittent electrical fault.
Something seems to be going on between Cura and the video system/OpenGL driver.