Ultimaker/Cura #19653: Material settings and ultimaker thing are hell.

patricsgamma, 2024-09-14:
### Cura Version

5.8.1

### Operating System

Osx

### Printer

Ultimaker 3

### Reproduction steps

# Material management in Cura is 💩 

Material management in Cura is a piece of frustration and your documentation is not helpful.

## Issue nr. 1, Material setup and sync:

So I have a custom ABS material that is set to print at 265°C
[image]

Your documentation states that I should sync this Material with my Ultimaker for it, to make it selectable.
According following documentation: https://support.makerbot.com/s/article/1667412729968

Now the thing is, none of these workflows are available to me and I can't see where they would be located in any menu...
Great documentation 💩 

Now this guide shows a Sync button that I just don't have, nice and helpful again.
https://support.makerbot.com/s/article/1667337567839

So issue number one is, I can't get my Material onto my Ultimaker 3 and I have no clue how to solve it.
Either fix the software or the documentation...

## Issue nr. 2, Print settings
 Values are not even consistent within Cura

Now when I go to **Print settings** for a part I want to print as follows
[image]

The print settings show now a default of 250°C when my Custom material is selected. 
Which makes no sense... Generic ABS is set to 230° and my Custom one to 265°, where the 250 default is coming from is beyond my comprehension. 
But of course I like to setup a Material, fail in sending it to my printer and then having to still configure it custom wise for every print, because why not.

As you can definitely read between the lines I'm currently not very tolerant towards anymore un-comprehensive guides and weird magic in software. So pls explain me like I'm 5 how to be able to use my material, KKTHX

### Actual results

🦄 💩 Read above

### Expected results

My material is available on my Ultimaker 3 and I can use it in print settings with configured values.

### Add your .zip and screenshots here ⬇️

The print just fails as my nozzle gets clogged when loading my material... likely due to the to low temperatures.
So there is no print file.

GregValiant, 2024-09-14:
Thanks for the report.
I'm not an Ultimaker guy but I played with this a bit.  I can't get it to work either.
It appears that "Default Print Temperature" never really changes it's value from whatever the "base material" is.
If I "Duplicate" a material then I am stuck with Default Print Temperature of the original material.
If I create a new material, then the problem seems to go away.

If you create a custom setting Profile and include the temperatures you want to use within it, then those temperatures will be used.
The Cura team will take a look.

patricsgamma, 2024-09-16:
Ok the material default behavior was a good hint, thank you.
As recommended in the documentation, I did not create a material from scratch but used the ABS default as a base, then tweeked the print and bed temperatures.

I still can't figure out why I can't export a material setting our of Cura and bring it to my Ultimaker 3.
My Cura doesn't show any of the menus to export, no matter over Digital Factory and network or USB.

Where does the magic trick to solve this, Anyone any knowledge on this?

patricsgamma, 2024-09-17:
The [documentation](https://support.makerbot.com/s/article/1667337567839) I mentioned above is talking about syncing materials through Digital Factory.
My printer is connected by network and I can access a Webpage called Digital Factory on my local network, which I assume the guide is talking about when talking about Digital Factory. So this step has happened.

Now creating new materials in Cura or Importing materials from following [guide](https://support.makerbot.com/s/article/1667410781024) does in no way give me any button or menu option to "sync material" and its also not happening automatically.

So my conclusion, as cynical it currently can be is: Nothing that is described in these guides is worth the 0.000x$ storage space they are saved on, as they just exist to confuse people that try to follow them.

patricsgamma, 2024-09-17:
My tone is not that I want to annoy anyone, but mostly an effect of being annoyed.
So Thanks for your help, as I have to admit before my first post, I didn't realize the printer generation part of the guide.

Still I miss to see the point where CURA as the software behaves interface wise differently on my printer then on others like S series printers.
So while I agree that my printer is not the newest and this is not a support forum, the slicer software is not making it clear to me what options are available, which I interpret as part of UX of this software.

So it is for example for me unclear where 1) Sync over USB, 2) Sync over local network and 3) Sync over Digital Factory are handled in Cura. 
As I understand now, only sync over local network is available to me, as I interpret out of above mentioned forum post.
Yet I have to find out anything about how this works.

patricsgamma, 2024-09-18:
The Ultimaker has OpenApi / swagger built in and I can get information like material from it, now we are starting to talk my love language ❤️

@HellAholic you just turned my hero for the moment, thank you!