Ultimaker/Cura #19655: Disable "USB Printing" plugin by default and add option to enable it in Add Printer wizard

Slashee-the-Cow, 2024-09-15:
### Is your feature request related to a problem?

Hiya! It's your Friendly Forum Assistant Moderator Who Doesn't Visit The GitHub Repo Much Slashee here.
There have been increasing numbers of people having trouble with models taking >15 minutes to load posting, and the solution always seems to be disabling the USB Printing plugin (and how many people find the solution and don't post about it... I don't have analytics so I can't tell you 🙂)
I think there's been another couple of issues which have had the same solution, but it's mostly the former.
I'm always quick to step in with the "USB printing is deprecated and no longer supported" spiel in most of these cases.
Sadly there's a lot of people who come to the forum with problems getting their printer to work over USB (or other problems and it comes up that they print over USB) so I know that there's still a seemingly not insignificant number of people who use it.

### Describe the solution you'd like

I think the USB Printing plugin should be disabled by default, but add an option in the Add Printer (in the non-networked section) wizard saying "I'll be printing over USB" which enables the plugin.
Maybe store it a machine instance setting so you can change it from the Machine Settings dialog without having to re-add a printer (then if you wanted to get all fancy like, you could dynamically enable it if any installed machines have it enabled).

### Describe alternatives you've considered

- Just having it disabled by default without a readily visible way to enable it would just send a bunch of people to the forums asking "why can't I print over USB?". 
- It could be no longer included in Cura but available in the Marketplace, but that would cause the above mentioned problem of beginners needing help finding it.
- I might love the idea of deleting it entirely, but there's still a few people running aged printers, although it might not be completely unreasonable to say "if you want to use this deprecated feature, you need version <x> or lower". Still probably not really an option.

### Affected users and/or printers

This would help prevent the "models take ages to load" problem and possibly other bugs for people who would never use USB printing anyway. 
Might shave a fraction of a second off the loading time for most people, too... not as important, although I have seen people complaining about the loading time.

GregValiant, 2024-09-15:
Hi Slashee.  This is a duplicate of #19373 and there is a response there by @wawanbreton .
That was a couple of months ago.  Maybe the team has had more discussion since then.

My preference would be having a control available to enable or disable from within Cura rather than having to go to the MarketPlace, enable/disable, and do a restart.  My impression is that approach would be a lot of work.

In addition, it would be nice to have the ability to have the printer responses load to a textbox on the Monitor screen.  Being able to send an M503 or run an M303 PID AutoTune and be able to see the printer response would be a plus and the Monitor screen appears to be under-utilized.  It would add functionality to what is now a one-trick-pony that is on its way to the glue factory.

wawanbreton, 2024-09-16:
Hi,
We had some discussion about this indeed, sorry I forgot to write a feedback.

According to our statistics, there are actually many users that use this feature. So if we turn it off by default, all the new users would just go into the situation of "Cura doesn't detect my printer" and it would be very not intuitive for them to think that the USB printing is something that is not enabled by default. We would then end up with many issues where the answer is just "enable the USB printing plugin", as @Slashee-the-Cow mentioned. Maybe some option to enable it easily would be preferable indeed, but currently Cura is not designed to enable plugins on-the-fly, so I don't know how good we can actually implement this.

It is also true that this USB detection mostly concerns "old" printers, more recent ones may be properly detected by VID/PID, and no baudrate detection should be necessary, in which case you can make something that works flawlessly in all cases, excluding support of old printers. So maybe we could keep the functionality in an external plugin for old printers, and have the modern ones work out-of-the-box. That is very speculative though :smiley: 

I'll try to have a new discussion with the team about this, as we haven't provided a proper answer and this is still an annoying issue.

Slashee-the-Cow, 2025-12-09:
> @Slashee-the-Cow <https://github.com/Slashee-the-Cow> can this be closed??

Yeah. What we got might not be quite as elegant as my idea but it does the
job.