Ultimaker/Cura #19656: It worked and then it didn't.

Dicealope, 2024-09-16:
### Cura Version

5.8.1

### Operating System

Windows 11

### Printer

Anycubic Predator

### Name abnormal settings

Support z Distance from .1 to .4

### Describe model location

No

### Describe your model

Printed 2cm smaller just fine, now it wont slice at all.

### Add your .zip here ⬇️

[Kemono Bunny Fursuit Base - 5351455.zip](https://github.com/user-attachments/files/17008375/Kemono.Bunny.Fursuit.Base.-.5351455.zip)

I printed this at 190mm just fine with .4 support distance on Z. but now it wont print at all, no matter what I change.

GregValiant, 2024-09-16:
Thanks for the report.
Without a project report and a log file there is no way to tell what might be going on with your installation.  I didn't have any trouble slicing it in 5.8.0 or 5.8.1.
This is with 5.8.1.
[image]

Dicealope, 2024-09-16:
Cura Version: 4.8
Operating System: Windows 10
Language: en_US
Machine Type: Ultimaker S5
Intent Profile: Default
Quality Profile: Fast
Using Custom Settings: No
Is Logged In: Yes
Organization ID (if any): ABCDefGHIjKlMNOpQrSTUvYxWZ0-1234567890abcDE=
Slice ID: aBcDeF01-2345-6789-aBcD-eF0123456789
Subscriptions (if any): 
Level: 10, Type: Enterprise, Plan: Basic 
Extruder 1: 
Material Type: PLA 
Print Core: AA 0.4 
Material Used: 1240 mm 
Extruder 2: 
Material Type: PVA 
Print Core: BB 0.4 
Material Used: 432 mm 
Print Settings: 
Layer Height: 0.15 
Wall Line Count: 3 
Enable Retraction: no 
Infill Density: 20% 
Infill Pattern: triangles 
Gradual Infill Steps: 0 
Printing Temperature: 220 °C 
Generate Support: yes 
Support Extruder: 1 
Build Plate Adhesion Type: brim 
Enable Prime Tower: yes 
Print Sequence: All at once 
... 
Model Information: 
Model 1 
Hash: b72789b9b... 
Transformation: [transformation matrix] 
Bounding Box: [minimum x, y, z; maximum x, y, z] 
Is Helper Mesh: no 
Helper Mesh Type: support mesh 
File type: STL 
Print Times: 
Infill: 61200 sec. 
Support: 25480 sec. 
Travel: 6224 sec. 
Walls: 10225 sec. 
Total: 103129 sec. 
Engine Statistics: 
Is Pre-Sliced: no 
Pre-Process Time: 7 sec. 
Slicing Time: 69 sec.

Dicealope, 2024-09-16:
>    1. Load the model in Cura.
>    2. Set up the slice. Slice the model.
>
> *Whether slicing was successful or not:*
>
>    1. Use the "File | Save Project" command to save a 3mf project file.
>    It will contain your printer, all your settings, and the model.
>    2. Zip the 3mf project file.
>    3. Use the "Help | Show Configuration Folder" command and locate the
>    Cura.log file.
>    4. Either add the cura.log file to the zip file containing the 3mf
>    project file, or create a second zip folder for the log file.
>    5. Post the zip folder(s) here.
>
> Those two files (project and log) may contain the necessary information to
> see what has happened. Without them the situation cannot be recreated. The
> model slices fine on my computer when I set Cura up with my normal
> settings. We need to know how you have set it up. That information will be
> in the Project File.
> We also need to know what notes Cura added to the log file if slicing
> failed, or if it didn't fail.

Hope this helps

GregValiant, 2024-09-16:
You must post the files directly to Github (their rules).  Email does not work.

Dicealope, 2024-09-16:
K

Dicealope, 2024-09-16:
[AP_bunny.zip](https://github.com/user-attachments/files/17019113/AP_bunny.zip)

GregValiant, 2024-09-16:
I still didn't have any trouble with it.  It's a 39 hour print.
[image]

The log file you posted stops abruptly.  Someone from the Cura team will need to take a look as their have more experience at reading the logs than I do.

Dicealope, 2024-09-16:
K, thanks.

closed 2024-09-16 by Dicealope