Ultimaker/Cura #19657: Scaling down of move/rotate axis lines fails if object is scaled up

bmateev, 2024-09-16:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

it doesn't matter whuch one I use

### Reproduction steps

When zoomed in on a object in Prepare view and move or rotate tool selected the axes lines (arrows) are outside of the screen. When zooming out the object is scaled down properly but the axes lines and arrows are not scaled down. 

### Actual results

Zooming out does not cause the axes to be scaled down and brought back in the visible portion of the screen.

### Expected results

The only way I found to bring the axes and arrows back is through clicking on the "Maximize\Restore down" title button in the top right corner of Cura's title bar,

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]

bmateev, 2024-09-16:
Just an FYI: This bug has been all along in all versions.

GregValiant, 2024-09-16:
Thanks for the report.
I've been doing triage here for about 3 years and I haven't seen a report on this behavior.

Someone from the Cura team will be along to take a look.  This might be just your system.  What is the resolution of your screen/monitor?  Everything looks small so I'm assuming a high resolution.

I am zoomed in quite a bit here and the movement icons remain "scaled to screen".  They remain the same relative size whether I zoom in or zoom out.
[image]

[image]

bmateev, 2024-09-17:
That is correct. The movement icons remain the same but when it bugs out is when they get scaled to the object zoom and get out of scope. Nothing to do with my resolution. It has something to do with going to the Monitor screen when printing through USB and getting back to the Prepare or Preview screen. Again, that does not happen all the time. I can't reproduce it at a commands notice. It is sporadic hence an annoying bug.

It is just an annoying bug that I thought you guys might want to look into. If there's a way for me to enable some debug trace or logging that might help in troubleshooting I am open to do so.

GregValiant, 2024-09-17:
In the MarketPlace is a plugin for "Extensive Support Logging" that you can install.  I don't know if it will catch what is going on there.  It might have some value.  It will add lines to the CURA.LOG file in your configuration folder.

bmateev, 2024-10-03:
I think I got the steps to reproduce. After slicing go to preview and select the object. Then click on "USB Print". Once it switches to Monitor screen click back on "Preview". That's when the movement icons start growing out of the screen.

GregValiant, 2024-10-06:
That didn't work for me.  No matter what I have done, the handles stay "scaled to screen" rather that being scaled to the model.
This might be a problem specific to your video sub-system.  A log file might have some value.

bmateev, 2024-10-15:
I installed the plugin but it does not show in the extensions. How do I find where the log files are being dumped? There is no information on the website.

bmateev, 2024-10-16:
It keeps happening but I don't know where to find the extensive logging files