Ultimaker/Cura #19658: support diagonal lines are stepped and not smooth.

FPVseb, 2024-09-16:
### Cura Version

5.5.0

### Operating System

macOS

### Printer

Voron V0.2

### Reproduction steps

i use the triangle support structure with conical support to support large overhanging areas. In diagonal lines of the support, they have steps instead if being smooth.

[image]

[image]

### Actual results

 this makes the lines have steps which causes the motor to shake when trying to print the line, and reduces the sprint speed.

### Expected results

Lines of the support should all be smooth.

### Add your .zip and screenshots here ⬇️

[image]

[image]

[V0_FillerBaseSource-66mm.3mf.zip](https://github.com/user-attachments/files/17014397/V0_FillerBaseSource-66mm.3mf.zip)

GregValiant, 2024-09-16:
Thanks for the report.
This appears to be a duplicate of #18045 .  The problem still exists in more recent versions of Cura.  I think this is on the Cura Team's list of things-to-do.
I'll mark this as a duplicate and maybe this will be pushed up on the list.  It certainly will slow down a print and shake up the printer.

GregValiant, 2024-09-17:
Here is an example of one of my prints.  The model is straight.  The infill has a lot of jig-jags.
[image]

FPVseb, 2024-09-29:
Hello,Sorry for my recent inactivity, I was very busy these past weeks.I have upgraded to Cura 5.8.1 but as you said, the stepped diagonal lines on the support didn’t go away. I played a bit with support settings and other dependencies and it seems that, whatever I change, the stepped diagonal lines always appear when i activate “conical support” (doesn’t matter the angle, support infill, support wall or anything).I hope this helps, the fact that I can reliably reproduce this issue is already a good step.Tell me if there is any more info, pictures or files I can provide. Looking forward to this bugfix.Thank you !Sebastien SchmidlinE-mail: ***@***.***

image.png (view on web)