Ultimaker/Cura #19659: Bridging Direction Issue

Andreuha, 2024-09-16:
### Cura Version

5.8.0

### Operating System

Windows 10

### Printer

Snapmaker J1

### Reproduction steps

This seems to overlap with [10372](https://github.com/Ultimaker/Cura/issues/10372), which is a very old entry now and lacking the example of reproduction attached below.

This is triggered by adjusting the wall settings to completely fill in the volume in the sections of the part that bridges are supposed to be connecting. 

### Actual results

After increasing wall count, cura orients bridges along the long axis of the gap, with one end of the bridge being suspended in air. 

I have found no way to get Cura to orient the bridges correctly without reducing wall count or switching to a stock profile (that is not dialed in for this combination of filament and printer)

[image]

### Expected results

Bridges should always go along the shortest direction of the gap. Bridges should generally avoid having one end be touching any unsupported material. 

Additionally, it would help a lot if there were a simple setting to just override the algorithm and force a specific direction for any bridges over a certain length layers across the part.

### Add your .zip and screenshots here ⬇️

[examples of same model with bridges in wrong and right orientation.zip](https://github.com/user-attachments/files/17017728/examples.of.same.model.with.bridges.in.wrong.and.right.orientation.zip)

GregValiant, 2024-09-16:
Thanks for the report.
When I open your "wrong orientation" file and slice it, I get this:
[image]

I think the settings are different as you have a lot more walls in your preview.  Here I have set it to 5 walls and it still looks OK.
[image]

If I make the wall count 5 it still doesn't match what you show in your screenshot.
[image]

Load the model and set Cura up.  Use the regular "File | Save Project" command and post the new 3mf file.

tremitry-darkstar, 2024-09-27:
I've experienced something similar. Community thread here (which includes my project file and a moderator comment that attempts to troubleshoot the issue: https://community.ultimaker.com/topic/44977-enable-bridge-settings/?do=findComment&comment=337582

In short, I have an Ender 3 that I'm trying to print some rail covers for. When I enable bridging mode, Cura attempts to print the first layer of the bridge along the long axis of the rails, which itself are just bridges over thin air. It seems to ignore the sides which a) exist and b) are a shorter distance to bridge.

Apparently, increasing the horizontal expansion to 0.2mm convinces Cura to print the first layer of bridging across the shorter axis (I believe 0.18mm is the point where Cura changes the axis it lays down the first bridging layer, but for my particular models, it did something weird to some of the circular parts of the models because it could not properly print a line around some of the rounded parts.) I'm literally printing these now to try it out.

closed 2024-10-02 by github-actions[bot]

tremitry-darkstar, 2024-10-03:
Do I need to recreate this as a new issue now? I'm trying to avoid making duplicate issues....