Ultimaker/Cura #19667: Constant crashing on Mac maybe after Sequoia 15.0 update?

HankLumbard, 2024-09-20:
### Cura Version

5.8.1

### Operating System

Mac Sequoia

### Printer

Ender 3 v2 Neo

### Reproduction steps

Sometimes as I'm moving models around other times completely random. Just closes and tells me unexpectedly quit.

### Actual results

Cura closes

### Expected results

Cura should not crash

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/user-attachments/files/17080556/cura.log)

GregValiant, 2024-09-21:
Thanks for the report.
You are the new record holder for "Most errors and warnings in a log file".
This one is in there a lot:
2024-09-20 16:31:43,700 - WARNING - [MainThread] cura.OAuth2.AuthorizationService.process_auth_data [194]: Failed to get a new access token from the server, retrying later.
2024-09-20 16:31:44,757 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [67]: Refreshing the access token for [https://oauth.onshape.com/oauth]
2024-09-20 16:31:45,149 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [392]:

This is a wild guess but it's something to look at:
If you have another USB device connected to the computer (headphones, blue-tooth device, etc.) and it is configured as a serial device, then Cura will try to connect to it.  It keeps trying because it IS a serial device and it keeps failing because it is not actually a printer and so the device never sends back the correct response.

Disconnect USB devices from the computer (not the keyboard or mouse) and then start Cura.  Go to the MarketPlace and disable "USB Printing".

This was also in the log file:
**2024-09-20 16:27:03,954 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.BLTH if baud rate 115200 works. Retry nr: 1. Wait timeout: 15
2024-09-20 16:27:03,954 - WARNING - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [45]: Unable to create serial connection to None with baud rate 115200
2024-09-20 16:27:03,955 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.BLTH if baud rate 250000 works. Retry nr: 1. Wait timeout: 15
2024-09-20 16:27:03,955 - WARNING - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [45]: Unable to create serial connection to None with baud rate 250000**

So Cura was attempting to communicate with something and was never successful.  Then those other errors started.
Maybe @wawanbreton can take a look at this?

wawanbreton, 2024-09-23:
I would also start by disabling the USB printing plugin if possible. We know it causes troubles, so having it disabled will make Cura less crash-prone (and more responsive) anyway.

If that doesn't help, can you try older versions of Cura ? Maybe 5.3 which has an different version of PyQt.

Also, usual questions:
* Did it start recently ? Following a Cura upgrade ?
* Did you recently install any plugin ?

HankLumbard, 2024-09-23:
Ok, so I disabled the detailed setting description plugin that I had recently added and it seems to be working fine now. Maybe the large pop ups on the hover over were taking too much memory? That's a complete guess but no more crashing! @GregValiant those connection errors are strange. I don't have any USB cables hooked up. I am running octoprint wirelessly, could just be testing different baud rates (?) as part of the connection sequence.

wawanbreton, 2024-09-23:
Good to know that you have an improvement :)

> Maybe the large pop ups on the hover over were taking too much memory?

I heard some strange stories about dialogs with Qt on MacOSm so that might be something related to it.

> I don't have any USB cables hooked up

It is a common case that you have internal hardware devices linked to your CPU by a USB connection, so even with no external device, Cura will see some USB ports, and try to connect to it relentlessly.