Ultimaker/Cura #19668: code in the start of program

Ted5855, 2024-09-21:
### Cura Version

5.8.0

### Operating System

windows 10

### Printer

Tronxy with 2 colors

### Reproduction steps

I slice the part then when i put it in my printer it crashed. Open up the program in word I saw all of this code in the beginning. I remove it all down to were it says the thumbnail begin  and it worked good. on the 5.7.1 it was not there.
[{TR}_cork_coaster_with_name.txt](https://github.com/user-attachments/files/17084185/TR._cork_coaster_with_name.txt)

### Actual results

the old rev worked

### Expected results

not crashed

### Add your .zip and screenshots here ⬇️

[{TR}_cork_coaster_with_name-v4.zip](https://github.com/user-attachments/files/17084197/TR._cork_coaster_with_name-v4.zip)

GregValiant, 2024-09-21:
Thanks for the report.
I found this line in the Gcode (line 24 right before "; thumbnail begin 32 32 1292":

**;Thumbnail generated by the Elegoo Neptune Thumbnails plugin version 4.4.0 (https://github.com/Molodos/ElegooNeptuneThumbnails)**

It would appear that your TronXY printer isn't playing nice with the Elegoo post-processor.  I suggest you remove it from the list of active post-processors.

That plugin is included with the Elegoo version of Cura.  I'll go ahead and close this as it is not an UltiMaker Cura problem.  I hope you understand.

closed 2024-09-21 by GregValiant

Ted5855, 2024-09-23:
I thank you so much, it work just fine. while I have you I was just wounding if you guys ever watch your program run. I mean like for my printer TronXY. when I use the 2 colors and it makes the tower, it sets for a long time, goes to the tower does it's thing goes back to the print then goes back to the tower then back to the print. To me it looks like a lot of time wasted. If you like I can send you one of my programs to look at and maybe tell me it is what I am doing. Thanks 

    

        Bill Goldy