Ultimaker/Cura #19679: Multiple Issues on Cura with Ender 3 V3 KE

Audyd1988, 2024-09-24:
### Cura Version

5.8.1

### Operating System

Windows 11

### Printer

Creality Ender 3 V3 KE

### Reproduction steps

1. When running an item that has a hole on the bottom where a part will be going into, it fills the hole in like it has a raft or something, the hole should be there so the parts can be put together. Can not find what setting would be doing this so it is in the program somewhere. Needs this fixed so we can run a multipart item
2. When slicing items it does not give the correct time for a machine that can run at a speed of 250, it tells it as time as if it were running at 50, need to know how long it takes to run an item.
3. It keeps adding twice the amount of material onto the item sense it does not tell the correct time also, or there is a bug there some where

### Actual results

[CE3V3KE_Head.zip](https://github.com/user-attachments/files/17122224/CE3V3KE_Head.zip)

### Expected results

Would like these items fixed so we can judge or time and material correctly please
Could not get a file for the #2 above to large and would not zip down small enough
[ScreenShot Cura Screen.docx](https://github.com/user-attachments/files/17122227/ScreenShot.Cura.Screen.docx)

### Add your .zip and screenshots here ⬇️

[CE3V3KE_Head.zip](https://github.com/user-attachments/files/17122235/CE3V3KE_Head.zip)
[ScreenShot Cura Screen.docx](https://github.com/user-attachments/files/17122236/ScreenShot.Cura.Screen.docx)

Asterchades, 2024-09-25:
A Gcode file is not useful for diagnostics in this case. Since this is already sliced there's no way to determine what may or may not have gone wrong as there's nothing relevant to inspect. You need to provide a Project file for any meaningful diagnostics - at present I can only assume the model itself is faulty and the "hole" is being defined as an un-capped solid (which will still slice as if it were a full solid), or you have the **Remove All Holes** mesh fix enabled and it's doing exactly what it says it's going to do.

Print time estimates depend on Cura having the correct information about your printer. They'll never be 100% accurate (especially on printers that don't use Jerk, rather Junction Deviation or Square Corner Velocity) but they should be close. However _all_ of the current Creality definitions all assume that the printer is configured for just 500mm/s² acceleration, which will have a big impact on a printer like the KE that is configured much higher. You need to manually provide Cura with more accurate information, which can be done one of three ways: manually editing the printer's definition file (not recommended), using the **Enable Acceleration Control** option and setting appropriate values, or using the **Printer Settings** plugin and assigning appropriate values to the Acceleration fields.

Material usage is not tied to the time estimate. It's calculated based on the sum of the E values that have been passed to the Gcode - something the slicer knows the exact values for - which is then fed through the user-adjustable density value to convert length into weight. Since the provided GCode file does not match the screenshot (please directly use an image in future, rather than embedding it in a document - it's far more portable, easier to view, and more efficient) there's no way to tell whether this estimate is correct or not. For something of that size, though, I wouldn't be surprised if it is accurate depending on your settings (which again would be present in a Project file).

Audyd1988, 2024-09-25:
Thank you, the first one worked I could not figure this issue out, I thought it was something that easy

#2 See the files attached the first one is an example of what I am talking about, on the Ender it should be about 16 hours to run and about 188 grams of material not 48 hours and 788 grams. The Second file is from or CR-10 to compare it to CR-10 = 40 hrs and 183 grams (I know the Times are different because of speed difference, but the grams are closer to what the original file I had ran).
I just know the time on and Ender 3 V3 KE should not say 48 hours and of course 788 grams for that item

[From CR-10 Crystal.zip](https://github.com/user-attachments/files/17131379/From.CR-10.Crystal.zip)
[From Ender Crystal.zip](https://github.com/user-attachments/files/17131380/From.Ender.Crystal.zip)

Asterchades, 2024-09-26:
Okay, yes - I think I see the problem here.

There appears to be at least five Infill extrusions that really, _really_ shouldn't be: two on layer 58, two on layer 59, and one more on layer 90. Just one of those on layer 58 (lines 956,692 and 956,694) is calling for over 40m (that's forty _metres_) of filament, followed by another 39m, in a bizarre attempt to run an extrusion to X1,547,075.666 Y1,495,591.156 (those are both around a kilometre and a half). Quite obviously that shouldn't be happening even once, never mind multiple times. This could easily account for the disparity in both time and filament (just the five obvious ones, if all the same, would be an extra ~400m of filament and an extra ~25km of movement), and would also result in a file that either won't print or print incorrectly at best, and at worst could easily result in a hot-end blob forming as 40m of filament is extruded in such a short distance twice in a row.

Setting the **Infill Density** to another value seems to remedy this, as does changing the **Infill Pattern**. For some reason this exact combination (30% Grid) with this particular model is causing some kind of error. So in the short term you can adjust either of these settings to print your models (and indeed for larger models with more sizeable voids I would recommend the "Cubic Subdivision" pattern to conserve material without losing strength - it doesn't help much on this dragon, but may help considerably on your lion).

Someone from the team will take a look.

Audyd1988, 2024-09-26:
I understand changing things might help, but this happens on more than one model and the combination is different so there is something in Cura for the Ender messing up. 
Also changing things does not help if this is what we need to be running our model at to get the quality and we need and want.
Also, that Infill pattern recommended does not seem to be a good one for most of our models we need strength in 90% of our models and that one seems a little flimsy.
So at least you were able to see the issue and are having people work on it.

I also have another question, is there a reason that Cura does not have the CR-10 Smart Pro as a machine choose. Cura finally got the Ender 3 V3 KE but still does not have the CR-10 Smart Pro?

Asterchades, 2024-09-27:
All 3rd party definitions are provided by the community, via pull requests here on GitHub. The reason is quite simply because someone submitted a definition for the V3 KE, while no-one ever did so for the Smart Pro.

The Cura team only maintains the Ultimaker (including Makerbot and the global default Custom FFF) definitions as that's what they have the appropriate hardware and experience for.