Ultimaker/Cura #19682: Cura initially not loading files upon import

gm094, 2024-09-25:
### Cura Version

5.8.1

### Operating System

Kubuntu 22.04 (x11 desktop with plasma 5.27 framework)

### Printer

All

### Reproduction steps

1. Launch Application Appimage through custom .desktop file (have used same file for years without issues)
2. import models by dragging file to build plate from another file window or by importing via the button built into Cura (same issue with both)

### Actual results

Models either don't load at all and I have to restart the application, or more commonly take a long time to load (upwards of 2 minutes for one file). The window is still responding, and I can use Cura (pan, change settings, open menus, etc) normally while waiting on a file to load in, the file not loading is the only issue. Once one file loads in, I can import other files quickly and normally without issue. This happens every time I try to use cura (currently about 30 times this month), and only became an issue after the last update. Only plugin I am running is octoprint integration, but that isn't the issue as far as I am aware

### Expected results

Simply drop a file and have it load in instantly. 

### Add your .zip and screenshots here ⬇️

No files since this happens with everything I try to import, regardless of type or printer or size.

MajliTech, 2024-10-06:
I also have the same issue, Plasma 6 on Fedora Linux 40,

MajliTech, 2024-10-06:
###### *H264 1280x720 30 fps, bitrate 1974 kb/s*

https://github.com/user-attachments/assets/30ba22ff-747a-4a70-9ac9-594a17b89759

Hopefully this shows what we are talking about. When minimizng the window, the object suddenly decided to load.

carloza, 2024-10-15:
Hi, I have the same problem since 3 or 4 mounth ago.
cura version: 5.8.1 (same issue in previous versions)
OS: Arch Linux
Kernel: 6.11.3-arch1-1
DE: plasma 6.2

MajliTech, 2024-11-28:
> Hi, I have the same problem since 3 or 4 mounth ago. cura version: 5.8.1 (same issue in previous versions) OS: Arch Linux Kernel: 6.11.3-arch1-1 DE: plasma 6.2

While it seems going nowhere, are you also by chance using Wayland with that Plasma?

gm094, 2024-11-28:
> > Hi, I have the same problem since 3 or 4 mounth ago. cura version: 5.8.1 (same issue in previous versions) OS: Arch Linux Kernel: 6.11.3-arch1-1 DE: plasma 6.2
> 
> 
> 
> While it seems going nowhere, are you also by chance using Wayland with that Plasma?

It's interesting that you mentioned that. I noticed that Cura doesn't launch at all when I'm using Wayland the error is something alone the lines of OpenGL 2.0 drivers not working (I can't remember the exact message off the top of my head.) The window launches without an issue on X11, however that is when the bug I outlined above consistently occurs.

On that note, the Cura window also freezes when opening project files in it. I have gotten in the habit of launching a project file about 5 minutes before I need it so it unfreezes when I'm ready, but this is still a very annoying workaround.

MajliTech, 2024-11-28:
I mention this because the guy I quoted uses Plasma, I use the same version with Wayland and this stupid bug occurs. Using with Flatpak.

gm094, 2024-11-28:
I will mention that I recently switched to endeavorOS (basically Arch with Plasma 6) and I have this same issue. But yes, something interfering with plasma seems to be the culprit. I am currently using the official app image from Ultimaker's website, I haven't tried flatpak yet but I'll make sure I try that soon.

carloza, 2024-12-09:
hi guys, sorry for replying too late (too much work and too much family), in short i use plasma with kwim on xorg (x11) and indeed i don't use wayland
@gm094 i also suspect that the problem is plasma related (or kwin idk).

didimau, 2025-01-14:
same problem with fedora 41 and kde
via drag and drop and file explore it takes a while to load, doesnt really matter how large the file is ... ~1min loading time
i can force to load it faster with opening the same file X times ... and i also get the model X times on my buildplate

strange behavior

mpiscaer, 2025-01-18:
I have the same issue I'm using Ubuntu 22.04 with Gnome as window manager.

I use the "official app image from Ultimaker's website"

MajliTech, 2025-01-18:
> I have the same issue I'm using Ubuntu 22.04 with Gnome as window manager.

That would make it not KDE's problem then.

didimau, 2025-01-18:
i found an old version 4.13 or so ... now i have no issues