Ultimaker/Cura #19684: Cura crashes when trying to open a file

alisabedard, 2024-09-25:
### Cura Version

5.8.1

### Operating System

"antiX-23.1 Arditi del Popolo

### Printer

Metal God

### Reproduction steps

I tried to open a file to slice with the file selection dialog.

### Actual results

 Cura crashed without showing a file selection dialog.

### Expected results

A file selection dialog should have appeared.

### Add your .zip and screenshots here ⬇️

[image]

Rio6, 2024-12-16:
I am experiencing the same issue on 5.9.0. My system is `Linux tomato 6.12.1-artix1-1 #1 SMP PREEMPT_DYNAMIC Fri, 22 Nov 2024 22:38:24 +0000 x86_64 GNU/Linux`

petr-cisar, 2024-12-18:
I tried to build 5.10 and it seems to suffer from a similar issue. This is written into the console log:

2024-12-18 15:33:30,644 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [448]: file:///opt/Cura/Cura/resources/qml/Cura.qml:624:5: QML FileDialog: Failed to load non-native FileDialog implementation:
qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml:4 module "Qt.labs.folderlistmodel" is not installed

Fatal Python error: Segmentation fault

Rio6, 2024-12-21:
Running with `QML_IMPORT_TRACE=1` gives
```
qt.qml.import: locateLocalQmldir: Qt.labs.folderlistmodel module's qmldir file not found
qt.qml.import: addLibraryImport: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml "Qt.labs.folderlistmodel" version ' invalid ' as ""
qt.qml.import: addLibraryImport: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml "QtQuick" version ' invalid ' as ""
qt.qml.import: importExtension: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml loaded ":/qt-project.org/imports/QtQuick/qmldir"
qt.qml.import: loading dependent import "QtQml" version 6.6 as ""
qt.qml.import: addLibraryImport: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml "QtQml" version ' 6.6 ' as ""
qt.qml.import: importExtension: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml loaded ":/qt-project.org/imports/QtQml/qmldir"
qt.qml.import: loading dependent import "QtQml.Base" version 6.6 as ""
qt.qml.import: addLibraryImport: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml "QtQml.Base" version ' 6.6 ' as ""
qt.qml.import: importExtension: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml loaded ":/qt-project.org/imports/QtQml/Base/qmldir"
qt.qml.import: loading dependent import "QtQml.Models" version 6.6 as ""
qt.qml.import: addLibraryImport: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml "QtQml.Models" version ' 6.6 ' as ""
qt.qml.import: importExtension: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml loaded ":/qt-project.org/imports/QtQml/Models/qmldir"
qt.qml.import: loading dependent import "QtQml.WorkerScript" version 6.6 as ""
qt.qml.import: addLibraryImport: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml "QtQml.WorkerScript" version ' 6.6 ' as ""
qt.qml.import: importExtension: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml loaded ":/qt-project.org/imports/QtQml/WorkerScript/qmldir"
qt.qml.import: addLibraryImport: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml "QtQuick.Controls" version ' invalid ' as ""
qt.qml.import: importExtension: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml loaded ":/qt-project.org/imports/QtQuick/Controls/qmldir"
qt.qml.import: loading dependent import "QtQuick.Controls.Fusion" version 6.6 as ""
qt.qml.import: addLibraryImport: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml "QtQuick.Controls.Fusion" version ' 6.6 ' as ""
qt.qml.import: importExtension: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml loaded ":/qt-project.org/imports/QtQuick/Controls/Fusion/qmldir"
qt.qml.import: loading dependent import "QtQuick.Controls.Basic" version 6.6 as ""
qt.qml.import: addLibraryImport: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml "QtQuick.Controls.Basic" version ' 6.6 ' as ""
qt.qml.import: importExtension: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml loaded ":/qt-project.org/imports/QtQuick/Controls/Basic/qmldir"
qt.qml.import: loading dependent import "QtQuick.Controls.impl" version 6.6 as ""
qt.qml.import: addLibraryImport: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml "QtQuick.Controls.impl" version ' 6.6 ' as ""
qt.qml.import: importExtension: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml loaded ":/qt-project.org/imports/QtQuick/Controls/impl/qmldir"
qt.qml.import: loading dependent import "QtQuick.Controls.Basic" version 6.6 as ""
qt.qml.import: addLibraryImport: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml "QtQuick.Controls.Basic" version ' 6.6 ' as ""
qt.qml.import: importExtension: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml loaded ":/qt-project.org/imports/QtQuick/Controls/Basic/qmldir"
qt.qml.import: loading dependent import "QtQuick.Controls.impl" version 6.6 as ""
qt.qml.import: addLibraryImport: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml "QtQuick.Controls.impl" version ' 6.6 ' as ""
qt.qml.import: importExtension: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml loaded ":/qt-project.org/imports/QtQuick/Controls/impl/qmldir"
qt.qml.import: locateLocalQmldir: QtQuick.Controls.Fusion module's qmldir found at "/tmp/.mount_UltiMauSCXUw/PyQt6/Qt6/qml/QtQuick/Controls/Fusion/qmldir"
qt.qml.import: locateLocalQmldir: QtQuick.Controls module's qmldir found at "/tmp/.mount_UltiMauSCXUw/PyQt6/Qt6/qml/QtQuick/Controls/qmldir"
qt.qml.import: addLibraryImport: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml "QtQuick.Controls.impl" version ' invalid ' as ""
qt.qml.import: importExtension: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml loaded ":/qt-project.org/imports/QtQuick/Controls/impl/qmldir"
qt.qml.import: addLibraryImport: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml "QtQuick.Controls.Fusion" version ' invalid ' as ""
qt.qml.import: importExtension: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml loaded ":/qt-project.org/imports/QtQuick/Controls/Fusion/qmldir"
qt.qml.import: loading dependent import "QtQuick.Controls.Basic" version 6.6 as ""
[... cut ...]
I tried copying the qml files into extracted AppImage `cp -r /usr/lib/qt6/qml/Qt/ PyQt6/Qt6/qml/`, which results in a different error
```
qt.qml.import: addLibraryImport: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml "Qt.labs.folderlistmodel" version ' invalid ' as ""
qt.qml.import: importExtension: qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml loaded "/tmp/squashfs-root/PyQt6/Qt6/qml/Qt/labs/folderlistmodel/qmldir"
qt.qml.import: locateLocalQmldir: Qt.labs.folderlistmodel module's qmldir found at "/tmp/squashfs-root/PyQt6/Qt6/qml/Qt/labs/folderlistmodel/qmldir"
```