Ultimaker/Cura #19685: Unnecessary travels from one object to another

gabrielforneck, 2024-09-26:
### Cura Version

5.8.1

### Operating System

Windows 11 (22631.4169)

### Printer

Ender 3 v1/v2

### Reproduction steps

1. Open my project
2. Slice

### Actual results

The program slices the project normally, but, at the layer 35, it makes some unnecessary travels from one object to another:
[image]

### Expected results

These travels are not needed.

### Add your .zip and screenshots here ⬇️

- The issue seems to be position sensitive: When I change the position of the objects, the issue doesn't occurs anymore.
- I don't know if it is a problem caused by a bad model.
- It seems to be a rare case. I would suggest a low priority on this case.

Models presented here are from [this remix](https://www.thingiverse.com/thing:4317411) on Thingiverse.

Here is a .zip file with all information I considered important to diagnosticate the problem that I am showing, including a video showing the step by step:
[bug.zip](https://github.com/user-attachments/files/17142266/bug.zip)

Asterchades, 2024-09-26:
This reminds me of an older bug that I'm currently unable to locate. The short of it is that the currently selected **Combing Mode** is creating the unnecessary travel paths - any mode other than "Within Infill" seems to work perfectly. From a purely outside point of view, it almost appears that it's crossing to a different model to find infill for the purposes of travelling, but I'm _fairly_ sure it's not quite that simple.

It's also worth mentioning that this model is not manifold. I'm not sure where exactly the issue is, but both the Mesh Tools plugin and 3D Builder take issue with the model as-is. While Mesh Tools is unable to repair it, 3D Builder is - and the repaired model does not present these errant travel moves. That of course doesn't excuse the behaviour, but may help in finding the root cause.

I'll give @GregValiant a ping on this. He perhaps has a link to the older report I'm thinking of, which will only help the diagnostic process if all such reports can be linked together.

In future you only need to provide a Project file that's in the state the bug occurs. This already includes all of your currently used settings and the model locations, which in this case are both important and slightly different to the provided Project file (though the included video did allow me to correct that).

_For QA: Import the file as a project, then clear the build plate. Import the file once more, this time as models so they get rearranged. This will produce the layout that causes the issue._

gabrielforneck, 2024-09-26:
Hi!
I started using projects to save my next prints a while ago, but I never realized that opening a project by choosing "Open as project" and "Import models" causes a different arrangement of the models on the baseplate. Here is an update of the .zip file where **project_v2.3mf** is a project that, when you import it as a project, leads to the problem.
[bug_v2.zip](https://github.com/user-attachments/files/17157080/bug_v2.zip)

Plus:
If I put the objects manually on the baseplate and put them in the same location as project_v2.3mf, the slicer creates unnecessary travels too. I also found that if you change the position very slightly, by something like 0.0001 or 0.001, the bug occurs with a different part of the object. I can make images or videos of that if you want.

gabrielforneck, 2024-09-26:
Another consideration:
The bug appears to be exclusive to 5.8.1 and 5.8.0, because loading the same project with version 5.7.0 doesn't cause the same problem.

Asterchades, 2024-09-27:
> I started using projects to save my next prints a while ago, but I never realized that opening a project by choosing "Open as project" and "Import models" causes a different arrangement of the models on the baseplate.

This is another _possible_ bug that's currently under investigation. Prior to Cura 5 the two would import identically, but ever since they behave differently unless specific conditions are met.

Appreciate the extra info. Little differences like that can help with diagnostics, as that helps narrow down the exact circumstances which result in the fault occurring.