Ultimaker/Cura #19692: Cura 5.8.0/5.8.1 crash on file open or file save dialog

jmccurley-lumco, 2024-09-27:
### Cura Version

5.8.1

### Operating System

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

1. Open Cura , as administrator or normally
2. Click the folder in the upper left, or File-> Open Files -> From Disk
OR
1. Open Cura
2. Drag and Drop a project file into the graphics window
3. Select open as project
4. Make any change to settings
5. Select File -> Save Project -> To Disk

### Actual results

File open dialog appears, but with the left-hand tree blank, or all text areas bright white instead of theme-color.  Then Cura crashes to desktop after a few seconds.

### Expected results

File dialog should fully appear and allow for file opening or saving

### Add your .zip and screenshots here ⬇️

[stderr.log](https://github.com/user-attachments/files/17166402/stderr.log)

jmccurley-lumco, 2024-10-01:
I have done a clean install of Cura and removed all old versions prior to the new install, no joy.  I also attempted to reset the settings for the Open dialog and common dialogs by removing the Registry Keys ComDlg32 and CIDOpen.  Interestingly enough, the crash still occurred, but it only re-created the ComDlg32 registry key, not the CIDOpen, implying that it never got far enough to even realize it was trying to use the open dialog, only that it was a dialog of some sort.

alisabedard, 2024-10-02:
This appears to be a duplicate of #19684, however with a different platform.

jmccurley-lumco, 2025-04-17:
Updating to Cura 5.10 seems to have corrected it for me.  I am now able to open and save normally.