Ultimaker/Cura #19706: Filament change script duplicates M600 command every time it gets send to printer or saved to disk

AndrejsSh, 2024-10-02:
### Cura Version

5.8.0

### Operating System

Windows 10

### Printer

Elegoo Neptune 4 pro

### Reproduction steps

1. Drop a model to buildplate
2. Go to Extensions-Post Processing-Modify G-Code
3. Add Filament change script
4. Set Layer = 2
5. Leave default values in other settings
6. Slice
7. Save G-Code to file
8. M600 command is added before layer #1
9. Save G-Code to file again
10. M600 command is added before layer #0 AND before layer #1
11. Save G-Code to file again
12. M600 is added again
13. Same thing happens when sending G-Code to printer - every time the M600 commands get duplicated

### Actual results

G-Code is generated with too many M600 commands

### Expected results

G-Code must be generated with single M600 at layer #1

### Add your .zip and screenshots here ⬇️

[EN4PRO_First_Layer_Patch-0.3mm.zip](https://github.com/user-attachments/files/17233790/EN4PRO_First_Layer_Patch-0.3mm.zip)

GregValiant, 2024-10-06:
Thanks for the report.
I cannot duplicate the problem with UltiMaker Cura 5.8.0.
I can duplicate the issue with the Elegoo fork of Cura.

Your project file does not have the "Create Thumbnail" post processor active, but there are both "Elegoo" Thumbnails" and "UM" thumbnails present in the file.  Problems have been noted when the Elegoo Thumbnail plugin when it is used with UltiMaker Cura.
From your gcode file:
**;Thumbnail generated by the Elegoo Neptune Thumbnails plugin version 4.4.0 (https://github.com/Molodos/ElegooNeptuneThumbnails)**
The Elegoo fork of Cura starts the gcode layer numbering with "LAYER:1" while Ultimaker Cura starts the layer numbering with "LAYER:0".  If the Elegoo post processor was not re-written to account for that it could be causing the problem.

The problem could be occurring because your model is 0.30mm thick but your layer height is only 0.20mm.  That means there will only be a single layer and so asking for a filament change on layer 2 (when there is no layer 2) could be a problem with Elegoo Cura.

Looking through the code of the UltiMaker "Filament Change" post processor, I don't see anything that would cause a double insertion, or an insertion at the wrong layer when used with UltiMaker Cura.

I'm going to remove the bug label but I'll leave this open so you can respond.  Because the Elegoo Thumbnail is showing up in the gcode I'm not sure which fork of Cura you are using.  This might be an Elegoo problem and this is the UltiMaker Cura Github site.

AndrejsSh, 2024-10-09:
Hi! I am using UltiMaker Cura 5.8.0 and i installed the Elegoo thumbnail plugin from Marketplace. I tried with a larger model and freshly installed Cura 5.8.0 and was able to repeat the problem. I will attach zipped project and gcode file. If it helps i can make a video?
[EN4PRO_100mXYZCalibrationV1.1(12_4_22).zip](https://github.com/user-attachments/files/17315417/EN4PRO_100mXYZCalibrationV1.1.12_4_22.zip)