Ultimaker/Cura #19712: Cura Not closing exit. force quit required

FPcrab, 2024-10-05:
### Cura Version

5.8.1

### Operating System

MacOS 15.0 (24A335)

### Printer

Tronxy XY2 PRO

### Reproduction steps

I closed the program and it freezes. have to force quit every time I quit cura

### Actual results

I go on and use my Mac

### Expected results

For cura to shut down

### Add your .zip and screenshots here ⬇️

[cura.zip](https://github.com/user-attachments/files/17267486/cura.zip)

log

FPcrab, 2024-10-05:
ARM M1 mac

cypher2001, 2025-01-21:
I'm having the same issue. Reinstall does not correct it.  Do we know what the issue is?

FPcrab, 2025-01-21:
> I'm having the same issue. Reinstall does not correct it. Do we know what the issue is?

i think it was some macos ussie. After upgraade macos to 15.2 (24C101) - problem was solved for me.