Ultimaker/Cura #19713: Tree Support Infill Printing In Air

GamesOnPaper, 2024-10-06:
### Cura Version

5.8.1

### Operating System

Windows 10

### Printer

Snapmaker J1 IDEX

### Reproduction steps

I have a Snapmaker J1 IDEX, I was on an older version and I upgraded to 5.8.1 and there is a bug where the layer in the support between the support and support interface prints the support infill across gaps. This only occurs with tree supports and not with normal supports. If I disable support infill it goes away, but that is not ideal. Attached is 3mf

### Actual results

[image]
This is what it currently looks like in 5.8.0 and 5.8.1

### Expected results

[image]
This is what it looked like in 5.7.1

### Add your .zip and screenshots here ⬇️

[SJ1_Support Test.zip](https://github.com/user-attachments/files/17267696/SJ1_Support.Test.zip)

LilBub, 2024-10-07:
Can you elaborate on what setting the support density to 0 will do?  That's a setting I typically set to 3 or 4%

GamesOnPaper, 2025-06-07:
Hello GregValiant,

I updated to Cura 5.10.1, and the bug is still persistent.
I cannot have a support density of 0% as I need support infill to keep taller trees from tilting or wobbling.

[image]