Ultimaker/Cura #19714: Failed Slicing on orizontal planes.

Andriuuu1901, 2024-10-06:
### Cura Version

5.8.1 - 5.7.1

### Operating System

Windows 11

### Printer

Elegoo Neptune 3Pro

### Reproduction steps

I imported models both made by me and from the internet and I started slicing.

### Actual results

The bases, or horizontal planes to the plane, are not calculated and in slicing holes are created at the base leaving the fill exposed.
If I turn the object the plane is correct, but if it's vertical position is not. It does this also with previous versions and other works. This was just a try for Halloween

### Expected results

I expected a closed print, but on the base or other orizonal planes is everything opened.

### Add your .zip and screenshots here ⬇️

[image]
[image]
[EN3PRO_WitchyBoo.zip](https://github.com/user-attachments/files/17269173/EN3PRO_WitchyBoo.zip)

GregValiant, 2024-10-06:
Thanks for the report.
You have all the "Top/Bottom" settings set to "0" so it would appear that the slice is exactly the way you have it configured.
[image]

This is with 4 bottoms and 4 tops.
[image]

I'll remove the bug label and close this as it appears to be a settings issue.  Let me know if you feel that is premature.

closed 2024-10-06 by GregValiant

Andriuuu1901, 2024-10-06:
Fixed!
Can you believe after ten days of trouble and posting this I literally and
casually opened that tab that I always keep close and find out exactly what
you're saying?
I set 1.26 mm, do you think it might be enough or are you suggesting to
raise it to 4? What does it change?

Thank you very much !

Andrea

GregValiant, 2024-10-07:
When I finish setting up any project, I save the work and then re-select the profile I used.  Cura will ask if I want to save the changes I made and I select "No".  That resets my profile to it's default and so when I start the next project, I know where I am.  For example, if I happened to use Spiralize for the previous project, it will be reset to "Off" and so I won't be looking at the new preview and going "WTF?".  Every model will require changes from my base profile.  That's why I only have 3 custom profiles.  One of them will provide a basis to get me started and then I go through all the settings and customize them for the current project.  It's better to take the time to get the setup right rather than wasting filament having to abort a print because you forgot something like moving the Z-seam to a different side.

You can set the Top/Bottom Thickness either in mm or in Number of Skins.  I just set the number rather than doing the math to figure out how thick the Top/Bottom will be.  Cura will always convert the thickness to an integer anyway.  You can't have 5 1/2 floors on the bottom.  If you have the Top/Bottom Thickness at 1.1mm and the Layer Height is 0.2mm then Cura would put in 5 Tops and 5 Bottoms because it can't hit .11 exactly.  Also consider that in the same situation the Initial Layer Height can make a difference as well.

The same sort of rule applies to Wall Thickness and Wall Count.  They are related but Wall Count will override the entry in the Thickness box.

A rule of thumb is a minimum of 3 walls and a minimum of 4 Top/Bottom skins.  Going less than those and you can get the infill leaving marks in the walls, or being able to see the shadow of the infill through the walls and top/bottoms.