Ultimaker/Cura #19720: support blocker failed

SimonH99, 2024-10-07:
### Cura Version

5.7.1

### Operating System

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

I imported a file in cura. I placed it on the bed. I placed support blocker on the areas i don't want the support to be. I slice.

### Actual results

After the slicing process, on the areas where no support should be there is support generated. 

### Expected results

no support should be generated where the supoort blockers are.

### Add your .zip and screenshots here ⬇️

[support blocker doesn't work.zip](https://github.com/user-attachments/files/17279932/support.blocker.doesn.t.work.zip)
[image]
[image]

GregValiant, 2024-10-08:
Thanks for the report.
This looks like a settings problem.  With the "Support Horizontal Expansion" at 0.8, the support is hanging off of that ledge and so Cura is building "Support to support the supports".  Those want to go to the build plate and that's what you see.

You can delete the support blocker, set the Support Horizontal Expansion to 0.0 and you should see this:
[image]

I'll remove the bug label from this as it appears to be a setting issue but I'll leave it open so you can respond.

SimonH99, 2024-10-10:
The Support Horizontal Expansion is set by deffault to 0.8 so i didn't care a thought about it.
Now it is doing what it should.
Thank you very much.

__________________________

Hofer Simon
Mechanical Engineering
GKN Hydrogen Italy S.R.L.

+39 0474 862652 - direct dial
+39 349 9851386 - mobile

website<https://www.gknhydrogen.com> | ***@***.***> | map<http://marketing.langleyholdings.com/signatures/maps/230/gkn-hydrogen-italy-s-r-l> | address<http://marketing.langleyholdings.com/signatures/address/230/gkn-hydrogen-italy-s-r-l> | legal<http://marketing.langleyholdings.com/signatures/disclaimer/230/gkn-hydrogen-italy-s-r-l>

[http://marketing.langleyholdings.com/signatures/logos/gkn-logo.png]<https://www.gknhydrogen.com>

[http://marketing.langleyholdings.com/signatures/logos/social-icons/facebook-green.png] <https://www.facebook.com/GKNHydrogen>   [http://marketing.langleyholdings.com/signatures/logos/social-icons/linkedin-green.png]  <https://www.linkedin.com/company/gkn-hydrogen>   [http://marketing.langleyholdings.com/signatures/logos/social-icons/youtube-green.png]  <https://www.youtube.com/channel/UC60CMpzGkIbhmVX66rUjnhw>

A Langley Holdings<https://www.langleyholdings.com> Company

EXTERNAL EMAIL

image.png (view on web)<https://github.com/user-attachments/assets/8eba3e64-355c-4dbf-a4ce-bdd04056ff88>

This e-mail from GKN Hydrogen contains information which is confidential and any attachment may be privileged.

closed 2024-10-14 by fieldOfView