Ultimaker/Cura #19724: AppImage still fails to run in Wayland

phrxmd, 2024-10-08:
### Cura Version

5.8.1

### Operating System

Linux (OpenSuSE Tumbleweed), KDE Plasma 6.1.5 Wayland, Intel UHD620

### Printer

Creality Ender-5

### Reproduction steps

1. Download the AppImage, `chmod a+x`
2. Start the application

### Actual results

An error message pops up:  "Could not probe OpenGL. This program requires OpenGL 2.0 or higher."

[image]

The console output is inconclusive:
```
> ./UltiMaker-Cura-5.8.1-linux-X64.AppImage
qt.qpa.wayland: EGL not available

>
```

### Expected results

The application should start with the QT Wayland UI.

### Add your .zip and screenshots here ⬇️

This a followup on issue #10815, which is closed. The behaviour is different now, but the basic problem (inability to start in Wayland) is still there. Either this issue or #10815 should be closed.

phrxmd, 2024-10-08:
There are two workarounds:

**Simple workaround: Start the AppImage in XWayland**

`QT_QPA_PLATFORM=xcb ./UltiMaker-Cura-5.8.1-linux-X64.AppImage` will force Cura into using an X11 GUI under X Wayland. This is crude, but it works.

**Extended workaround: Modify the AppImage to include the Wayland libraries** 

Download AppImageTool from https://github.com/AppImage/appimagetool/releases, then issue the following:

```
./UltiMaker-Cura-5.8.1-linux-X64.AppImage --appimage-extract 
cd squashfs-root 
cp /usr/lib/libwayland-client.so.0.23.1 ./libwayland-client.so.0
cp /usr/lib/libwayland-egl.so.1.23.1 ./libwayland-egl.so.1 
mv com.ultimaker.cura.desktop com.ultimaker.cura.desktop.bak
sed "s/.png//" com.ultimaker.cura.desktop.bak > com.ultimaker.cura.desktop
cd ..
ARCH=x86_64 ./appimagetool-x86_64.AppImage squashfs-root 
mv UltiMaker_Cura-x86_64.AppImage UltiMaker-Cura-5.8.1-linux-X64-with-wayland.AppImage
```
This modified AppImage starts fine under Wayland for me.

It seems like this is a simple packaging error where the AppImage build fails to include Wayland client libraries.

smartavionics, 2024-10-08:
You should just be able to remove those libraries from the appimage and then the system libraries will be found instead. I think it is a mistake for those libraries to be included in the appimage. I have now removed them from my Cura releases along with quite a few other libraries that still get bundled in the UM releases.

phrxmd, 2024-10-08:
#10815 is reopened, let's continue the discussion there.

closed 2024-10-08 by phrxmd