Ultimaker/Cura #19727: fuzzy skin determination inconsistent

RobWickham, 2024-10-08:
### Cura Version

5.7.1

### Operating System

Windows 10

### Printer

Geeetech Mizar S

### Reproduction steps

Enabled fuzzy skin and inside only, then observed the inner surface of the model and it's facets and adjacent walls were being treated differently. A intentional bump on the inside wall will be fuzzy, while the wall it's on will be smooth.

### Actual results

[image]

### Expected results

 I expect anything in the inside will to be skipped when using inside only.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-10-14:
Thanks for the report.
Please post a project file.  I can't tell what is going on with the spots.  The geometry of the model is indicating that the vertical feature on the right has no "inside" to exclude so that one makes sense.

RobWickham, 2024-10-29:
[Lamp Cover.zip](https://github.com/user-attachments/files/17561175/Lamp.Cover.zip)

Project file now added, sorry for the delay.

GregValiant, 2024-10-30:
Thanks for the file.
What's happening is that Cura can't determine where the "outside ends and the inside starts".  This can be adjusted with "Support Blockers" configured as "Cutting Mesh" and with "Fuzzy Skin" turned off for the blockers.

Here is the tall tab.  The problem with this feature is that it only consists of "outside".
[image]

Here are the bumps on the other side.  They are not parts of the "inside" and so get "excluded from being excluded".
[image]

I'll remove the bug label from this and close it as the problem involves the model geometry, and there is a means to fix it.  I hope you understand.

Here is what I did.  Placement of any "Cutting Mesh" is critical or they can cause walls to have defects.
[GV Lamp Cover.zip](https://github.com/user-attachments/files/17571466/GV.Lamp.Cover.zip)

closed 2024-10-30 by GregValiant