Ultimaker/Cura #19731: brim adhesion improvement: will all internal area, not just outside

goofyseeker311, 2024-10-09:
### Is your feature request related to a problem?

if you have multiple co-centric objects printed at once that touch the print bed, without separate full supports,
then only the outer object gets the brim. raft would be overdone, and objects would lose the print bed contact.
so brim should be outwards from all object bed contact are point, also inside and between the contact points.

### Describe the solution you'd like

have a "full brim" adhesion option that extends a brim from all walls outwards and inwards. ie symmetric brim, not outwards only.
the picture below should have two additional brims, between the two object walls area, and the inside area.

### Describe alternatives you've considered

current brim and raft do not fit the best functionality.

### Affected users and/or printers

any printer, any user.

### Additional information & file uploads

current brim adhesion in a double object at-once print.
[image]

Asterchades, 2024-10-10:
Strange. Normally the complaints are that the *Brim* functions similar to the way that you've described that you want it to - not in the way that you've shown it currently behaving.

Check your *Brim Location* setting. You may find you can already get what you're after by using the *Everywhere* option, as opposed to the default *Outside Only* one. NB This setting is not visible until *Advanced* visibility unless you manually toggle its visibility, but can always be found via the setting search box.

goofyseeker311, 2024-10-10:
Oh so the functionality setting was already there. Everything is fine then.

goofyseeker311, 2024-10-11:
On the same settings discussion, why does cura set custom material printing temp to 220C, but not to 260C calculating it to 235C. Yep with two different custom materials.

closed 2024-10-11 by goofyseeker311

Asterchades, 2024-10-11:
That's going to vary based on what actual printer definition you're using. Some of them have overrides included for certain nozzle sizes and material types which can change the print temperature - sometimes even outright ignoring the *Material Properties* rather than simply adjusting them. In this case, the base Elegoo definition (which provides the profiles for the Neptune 3 Pro) sets PETG to 235°c/70°c, while ABS and ASA will be set to 240°c/80°c and TPU will be set to 230°c/30°c. PLA is almost left untouched, but when using a 0.2mm nozzle it will be set to 190°c, while a 0.8mm nozzle will be set to 15°c higher than specified in the *Material Properties*.

All of this is in addition to other settings various combinations of nozzle sizes and materials will make.

This isn't something that the Cura team has done - this is the result of the community pull request setting those values. The team, as much as is possible, prefers to be hands-off when it comes to community submissions as they have neither the time nor the appropriate hardware to perform a full validation. Provided it isn't going to break anything, either software-wise or hardware-wise, they'll leave it alone. Only the Ultimaker (including global default and now MakerBot) definitions are directly handled by the Cura team.

No matter what happens, the *Print Profile* - the list down the right hand side of the main UI - has the final say in all settings. What's listed there will be used, letting you override any and all of these questionable configurations.

By default (and this can also be changed by definitions) the maximum printing temperature is 365°c, with a warning (also configurable) at 285°c. For posterity the minimum is set to -273.15°c, and there's really no reason to ever change this as that's also absolute zero (ie 0K).

goofyseeker311, 2024-10-11:
moved discussion to actual bug issue.