Ultimaker/Cura #19734: CURA 5.8.1 start printing before heating the nozzle

WeiglN, 2024-10-10:
### Cura Version

5.8.1

### Operating System

Windows 10

### Printer

Ultimaker Original +

### Reproduction steps

I constructed a simple modell with Freecad and produced a .stl File. I transfered it to Cura 5.8.1 
[UMO_Vogelhaus-Revolution (Meshed).txt](https://github.com/user-attachments/files/17326602/UMO_Vogelhaus-Revolution.Meshed.txt)

### Actual results

The Printer starts normal with heating the bed, but when the 70° was reached printing starts.
In the G File the reason is obvious, the M 105 and M 109 is missing: 
;Generated with Cura_SteamEngine 5.8.1
M140 S70
M105
M190 S70
M104 S230
M82 ;absolute extrusion mode
I tried different an older Versions 5.7.0 and that works

### Expected results

Expected that the nozzle is heated before start printing

### Add your .zip and screenshots here ⬇️

[image]

LilBub, 2024-10-10:
Yes, this one has bit a few of us. 
If you get the Machine Settings plug in, you can check if these things are ticked

[image]

WeiglN, 2024-10-28:
same problem with Cura 5.9.0-beta
;TARGET_MACHINE.NAME:Ultimaker Original+
;Generated with Cura_SteamEngine 5.9.0-beta.1
M140 S70
M105
M190 S70
M104 S230
M82 ;absolute extrusion mode
G21 ;metric values

WeiglN, 2024-10-28:
it is ok with 5.7.1
;TARGET_MACHINE.NAME:Ultimaker Original+
;Generated with Cura_SteamEngine 5.7.1
M140 S70
M105
M190 S70
M104 S230
M105
M109 S230
M82 ;absolute extrusion mode
Is not ok with Cura 5.7.2
;TARGET_MACHINE.NAME:Ultimaker Original+
;Generated with Cura_SteamEngine 5.7.2
M140 S70
M105
M190 S70
M104 S230
M82 ;absolute extrusion mode