Ultimaker/Cura #19735: PI3MK3M_GHOST WRANGLER Annie Valentino.stl

BattlePig, 2024-10-10:
### Cura Version

5.1.0

### Operating System

Windows 11

### Printer

Prusa i3 MK3s

### Name abnormal settings

I attempted different infill settings with the same failed effect.

### Describe model location

I leave the model as opened and select supports.

### Describe your model

I had no trouble slicing this on any other printer (I have 6), and other Hero Forge files sliced fine with the same settings.

### Add your .zip here ⬇️

[PI3MK3M_GHOST WRANGLER Annie Valentino.zip](https://github.com/user-attachments/files/17328687/PI3MK3M_GHOST.WRANGLER.Annie.Valentino.zip)

GregValiant, 2024-10-14:
Thanks for the report.
Cura 5.1 is 10 or 12 versions old.  I suggest you try 5.8.0 (which did not have any trouble with the model) and see if it makes a difference.
I'll go ahead and close this as it is in regards to a 2 1/2 year old version but you can still respond here.  Let us know whether the newer version fixed the problem.

GregValiant, 2024-10-14:
That model has some errors in it.  Newer Cura versions are more forgiving but you should check your models for errors as a matter of course.  It can save a lot of time fooling with Cura settings when the issue is actually the model.

I'll go ahead and close this.

closed 2024-10-14 by GregValiant