Ultimaker/Cura #19736: materials default printing temperature caps at 235C (missing hardened steel nozzle settings)

goofyseeker311, 2024-10-11:
### Cura Version

5.8.1

### Operating System

Win11 RTX3060

### Printer

Elegoo Neptune 3 pro

### Reproduction steps

1. Create two custom materials (for hardened steel nozzle), other with default printing temperature at 220C and other 260C.
2. Activate both 220C and 260C profiles, observe calculated print settings printing temperature. should be exactly same as material default printing temperature.

### Actual results

220C material gives print settings printing temperature of 220C, and 260C material gives 235C.

### Expected results

why does cura set custom material printing temp to 220C, but not to 260C calculating it to 235C. Yep with two different custom materials. Is there a hard temp limit of 235C in the code. I can force the printing temp to 260C in the printing settings per material but that would defeat the material settings purpose. Also Elegoo Nepture 3 Pro has a limit of 260C of the nozzle temp, not 235C.

You dont have elevated temp hardened steel nozzle material profiles by default. Say 220C for PLA instead of default 205C for brass nozzle. Yes nozzle type drop down in the material nozzle drop down selection tab.

### Add your .zip and screenshots here ⬇️

[image]

fieldOfView, 2024-10-13:
The Elegoo base profiles for PETG specifically set the `default_material_print_temperature` to 235. These profiles override what is set in the material.

You could install the "Material Settings" plugin from the Marketplace. That will add an option in the context menu for the Print Settings menu that will let you use the value from the material (instead of from the profile).

goofyseeker311, 2024-10-13:
no, I just made a print setting that sets print temperature to 220C and 260C, for any selected print material.

fieldOfView, 2024-10-14:
That also works.

Anyway, though I agree that the UX is not great here, it is "as designed" and "in accordance with the printer profile you chose", so I will close this issue.

closed 2024-10-14 by fieldOfView

goofyseeker311, 2024-10-14:
maybe its better to be a feature request then. not a bug.

goofyseeker311, 2024-10-14:
make the program logic function with different nozzles, hardened steel nozzles, not just brass nozzles. whatever form that takes. with the current materials.