Ultimaker/Cura #19739: [5.8.1] Cura won't start on Ubuntu 24.10

csetera, 2024-10-11:
### Cura Version

5.8.1

### Operating System

Ubuntu 24.10

### Printer

Ender 3 Pro

### Reproduction steps

Attempted to start Cura Appimage on Ubuntu 24.10 and it failed to start.  Trying to launch from terminal shows the following error for me:

```
/tmp/.mount_ultimaD1nZbh/UltiMaker-Cura: symbol lookup error: /lib/x86_64-linux-gnu/libEGL_mesa.so.0: undefined symbol: wl_display_create_queue_with_name
```

### Actual results

Attempted to start Cura Appimage on Ubuntu 24.10 and it failed to start.  Trying to launch from terminal shows the following error for me:

```
/tmp/.mount_ultimaD1nZbh/UltiMaker-Cura: symbol lookup error: /lib/x86_64-linux-gnu/libEGL_mesa.so.0: undefined symbol: wl_display_create_queue_with_name
```

### Expected results

Appimage starts

### Add your .zip and screenshots here ⬇️

Should be covered above

mechano, 2024-10-12:
Same problem just discovered yesterday after upgrade to 24.10.

skandalfo, 2024-10-12:
Happens to me too. Same error message. Using an AMD graphics card if that helps.

csetera, 2024-10-12:
Good thought on the comment about graphics card.  I have the Intel/Nvidia graphics card.

mechano, 2024-10-12:
I've MSI Radeon RX 6750 XT using Wayland.

saintaardvark, 2024-10-12:
Seeing this problem on Debian trixie/sid with Wayland:
```
$ lspci  | grep -i VGA
00:02.0 VGA compatible controller: Intel Corporation WhiskeyLake-U GT2 [UHD Graphics 620] (rev 02)

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux trixie/sid
Release:	n/a
Codename:	trixie

$ uname -a
Linux melodia 6.9.7-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.9.7-1 (2024-06-27) x86_64 GNU/Linux
```

csetera, 2024-10-13:
I also thought I'd mention that I've been running with Wayland since before the 24.10 upgrade.  So, while this is clearly Wayland-related, something must have changed in Ubuntu 24.10 to trigger this.

EmilEmilchen, 2024-10-19:
Happens to me as well, just upgraded yesterday, wanted to print something but no luck! 
```
WAYLAND_DISPLAY="" UltiMaker-Cura-5.8.0-linux-X64.AppImage
```
worked like a charm, Thanks!

mechano, 2024-10-19:
Found the way to pass the ENV variable **WAYLAND_DISPLAY=""** to  the application icon into menu. 

I've **$HOME/.appimage** directory where I save all my applimages file. 
Into this directory Cura's app is named: **UltiMaker-Cura.AppImage**

I created the .desktop file: 

**$HOME/.local/share/applicationsUltimakerCura.desktop**

and put the Cura icon into: **$HOME/.local/share/icons/ultimaker-cura.png**

into this file copy this entries:

```
[Desktop Entry]
Name=UltimakerCura
GenericName=3D Printing Software
Icon=ultimaker-cura
Exec=env WAYLAND_DISPLAY="" /home/mechano/.appImage/UltiMaker-Cura.AppImage %F
Terminal=false
Type=Application
MimeType=model/stl;application/vnd.ms-3mfdocument;application/prs.wavefront-obj;application/x-amf;
Categories=Graphics;3DGraphics;Engineering;
Keywords=3D;Printing;Slicer;slice;3D;printer;convert;gcode;stl;obj;amf;SLA
StartupNotify=false
StartupWMClass=ultimaker-cura
```

You will have the icon into App->Graphics menu.

yesrod, 2024-11-20:
Fixed in Cura 5.9, at least on my machine.

laverdone, 2024-11-21:
> Fixed in Cura 5.9, at least on my machine.

Now cura ignore wayland session (export QT_QPA_PLATFORM=wayland) and will start always on Xwayland! This is an workarround not a solution. Too bad.

fliegentoeter, 2025-07-24:
I can run the 5.10.1 appimage but I cannot load any stl files in it...
Same for the workaround for 5.4.0 "env WAYLAND_DISPLAY="" /UltiMaker-Cura-5.4.0-linux-modern.AppImage 
It starts fine, but fails to load any files...

Still stuck on an Appimage of 4.13.0
Seems like it is time to switch unfortunately. Is there a way to import Profiles to some other slicer that works on linux?

I am on Manjaro stable

Daniel-Khodabakhsh, 2026-03-01:
I ran into this issue on 5.12 but I can run it in Wayland with these steps:

First, similar to what @laverdone mentioned above, extract the AppImage and delete `libwayland-client.so.0`:

```sh
UltiMaker-Cura-5.12.0-beta.1-linux-X64.AppImage --appimage-extract
rm squashfs-root/libwayland-client.so.0
```

no need to delete the other files.

Next step is make sure you have `mesa` and `wayland` packages (or similar packages) are installed in the environment.

Finally, launch the extracted application with `-platform wayland` flag:

```sh
squashfs-root/AppRun -platform wayland
```

rephlex, 2026-10-25:
That `libEGL_mesa.so.0: undefined symbol` is the AppImage picking up your system libEGL while carrying an incompatible one - a classic bundle mismatch. On Ubuntu there is a simpler path: my company publishes signed native .debs for 24.04 and 26.04 (Cura 5.13.0).

24.10 itself is end-of-life; if you are (or move) on 24.04 LTS:

```
curl -fsSL https://mirror.plustech.de/plustech.gpg | gpg --dearmor -o /usr/share/keyrings/plustech.gpg
echo "deb [signed-by=/usr/share/keyrings/plustech.gpg] https://mirror.plustech.de/ubuntu/ plustech-noble main" > /etc/apt/sources.list.d/plustech.list
apt-get update && apt-get install ultimaker-cura
```

https://mirror.plustech.de/ , community repo with published sources.