Ultimaker/Cura #19740: hot end resets to 175 when print starts

AxleJack, 2024-10-12:
### Cura Version

5.8.1

### Operating System

windows 10

### Printer

creality ender 2 pro

### Reproduction steps

almost every time i start a print the hot end temperature switches to 175. 
Its easy to fix as I manually turn it back up to 210.
But its a pain :)

### Actual results

prints ok

### Expected results

emmm..   prints ok

### Add your .zip and screenshots here ⬇️

ok

Asterchades, 2024-10-13:
Some printer definitions will initially pre-heat the tool to the *Material Standby Temperature* during the Start Gcode. When another event has occurred - such as the print bed reaching full temperature or ABL routines being executed - it will continue to heat the tool up to the full printing temperature without any further user intervention. The *Ender-3 Pro* is one such definition (it waits for the bed to heat) and there are doubtless others.

Without the information requested when submitting the issue there's no way to say for certain that this what you're encountering, but I have a strong suspicion it is based on the 175°c specified and all such previous reports of the same. A Project file would have very quickly shown what was occurring, as would a small excerpt from the resulting Gcode file (it wouldn't require more than 100 lines).

GregValiant, 2024-10-14:
I don't see an "Ender 2 Pro" as a printer option in Cura 5.8.1.  If your are pre-heating your hot end to print temperature, and your "StartUp Gcode" has a line like:
"M104 S{material_standby_temperature}"
then as Asterchades explained - it is telling the printer to cool down.  That is usually a good thing for printers equipped with ABL as it will keep the hot end from oozing during the leveling process.

So this is not a Cura bug but rather something that your printer definition is requesting.
Printer definitions for 2nd party printers are developed by either the manufacturer or by a 3rd party who owns one.

I'll go ahead and remove the bug label from this and close it.  If you feel that is premature then let us know and I can re-open it.  We'll need a project file to do any investigation.

closed 2024-10-14 by GregValiant