Ultimaker/Cura #19742: Blank Window

NobodysBiz987, 2024-10-12:
### Cura Version

5.8.1

### Operating System

MacOS

### Printer

Custom

### Reproduction steps

1. I opened Cura

### Actual results

The mystery window opens in the control bar. I don't remember what I did to get it at first, but it's been around for a while and several updates. If I close the main Cura window it disappears, so it seems to be somewhat entangled to the normal Cura window.

### Expected results

It should open the normal Cura window. 

### Add your .zip and screenshots here ⬇️

[image]
[image]

Sistav, 2025-04-27:
I have the exact same problem on v. 5.10.0 using MacOS Sequoia 15.4.1 on an M4 Mac Mini. When hovering over the blank window the tab icons do not appear, and no commands work on it, however you can drag it around. It opens alongside cura, but starts minimized in the task bar. When you close Cura, it also closes, however it seems to be in a different window than the main Cura Window. 

[image]

[image]

[image]

Splunkon, 2025-05-08:
I have the same problem on Cura 5.10.0 and macOS 15.4.1. 

[image]