Ultimaker/Cura #19743: Error message Installing plugin.

sjj54, 2024-10-13:
### Cura Version

5.8.1.

### Operating System

Windows 10 64 bits

### Printer

Ultimaker Cura Version:  5.8.1

### Reproduction steps

Error message Installing plugin.
How can I remove this error message, or the pulgin that causes this error message.

### Actual results

I uninstalled Cura and reinstalled it.
Removed all plugins.

### Expected results

Unfortunately the problem remains.

### Add your .zip and screenshots here ⬇️

[image]
[image]
[cura.log](https://github.com/user-attachments/files/17355470/cura.log)

GregValiant, 2024-10-14:
Thanks for the report.
I haven't seen this problem.  Someone from the Cura team will take a look.

From the posted log file:
2024-10-13 12:48:59,496 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [392]: request[db99eb5e][get][PyQt6.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')][timeout=10][no-data] got an QNetworkReplyError Error transferring https://api.ultimaker.com/cura-packages/v1/user/packages - server replied: . The server returned: b'{"errors": [{"id": "60587013-292e-4058-a1fa-e2bdf2d14bbc", "code": "tokenExpired", "http_status": "403", "title": "The given token has expired.", "detail": "Signature has expired", "meta": {"error_type": "AuthError", "url": "https://api.ultimaker.com/user/packages"}}]}'
2024-10-13 12:48:59,511 - WARNING - [MainThread] Marketplace.CloudSync.CloudPackageChecker._onUserPackagesRequestFinished [76]: Requesting user packages failed, response code 403 while trying to connect to PyQt6.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')

sjj54, 2024-10-18:
HelloWhen can I expect the solution to this problem because I have already tried everything.GreatingsSjors Fikkert Verstuurd vanaf mijn iPhone

GregValiant, 2024-10-19:
The Cura team prioritizes their workload and gets to things as they can on the basis of those priorities.
If a problem can be re-created it can be addressed and a fix may appear in the next release of Cura.

Plugins are stored in the "Configuration Folder" and the "plugins" sub-folder.  You could try searching there to see if the offending plugin is there and delete the plugin folder.

The "Native CAD import plugin" is only available if you have the "special subscription" to "Cura Enterprise Edition".  The warning may be coming up because of the way you installed Cura.  Maybe @HellAholic can shed some light on this.

sjj54, 2024-10-20:
Hello, I solved the problem myself. After deleting everything from Cura slicer several times, including the hidden files and cleaning the registry in Windows 10, the error message continued to appear after logging in to the Ultimate Cura account after synchronizing with the account after a new installation. If I didn't log in there was no problem. I decided to delete the account on Ultimate site and created a new account with a different email address and logged in there. Now the problem has been resolved and the error message has disappeared. Thank you very much for the help.Yours sincerely. SjorsVerstuurd vanaf mijn iPhone