Ultimaker/Cura #19744: Only basic settings visible

wasikan-git, 2024-10-13:
### Cura Version

5.8.1

### Operating System

Linux Mint

### Printer

Prusa i3 Mk3S

### Reproduction steps

1. Click Settings/Configure settings visibility
2. Select eg. Advanced

### Actual results

Only basic settings are visible in the settings panel, no matter what was selected (Advanced, Expert, All).

### Expected results

Different sets of settings (according to selection) should be visible in settings panel.

### Add your .zip and screenshots here ⬇️

[image]

Asterchades, 2024-10-14:
You need to switch to the Advanced/Custom page in order to view the individual settings, regardless of their visibility state. Simply click the *Show Custom* box at the top of the settings drop-down. A similar button should be present on the Custom page to change back to the Simple/Recommended view.

closed 2024-10-14 by fieldOfView

wasikan-git, 2024-10-14:
Thank you, everything works. Though, I wouldn't call it intuitive.