Ultimaker/Cura #19746: Adaptive Purge

hitnrun30, 2024-10-14:
### Is your feature request related to a problem?

Not really

### Describe the solution you'd like

The new way of getting the purge line is to do adaptive purging.  Instead of a line on the front or along the side an adaptive purge is near to where the first layer starts and will purge out a thick line of filament.  In essence saving time and filament and the bed since it wont be purging in the same place.

### Describe alternatives you've considered

If we could somehow get a settings value for the start gcode for the printer that could give first_layer_print_min and first_layer_print_max positions or first_layer_start_x and first_layer_start_y  we can offset those value to be just to the right of that and make a small thick purge line.

### Affected users and/or printers

All

### Additional information & file uploads

https://www.printables.com/model/1035759-adaptive-purge-for-any-3d-printer-using-slicer-var

LilBub, 2024-10-15:
I have to admit, I like how the Ultimaker's do that purge blob or, in the older Ultimakers, they extrude out about 100mm (or so) of material on the front left, then off they go.

I couldn't figure out how to make my non Ultimaker machines do that, so I have stuck to a small purge line like Greg's.

hitnrun30, 2024-10-15:
The problem is if the purge in the same location over and over again it can be detrimental to your surface. I've seen a lot of times where the material separates because of the heat and the running in the same exact place over and over again. Also doing this probably will save time.

GregValiant, 2024-10-15:
_"...it can be detrimental to your surface..."_
I don't know about that.  I've printed thousands of things on my 4 year old Ender, and I haven't had a problem.

I'm not knocking your suggestion, but it sort of sounds like a "skirt around the skirt".  There are settings for "Skirt/Brim Line Width" and "Skirt/Brim Flow" that could be adjusted as well.
Here I've made the skirt 5 loops with the flow at 150% and the line width .6 on a 0.4 nozzle.

[image]

hitnrun30, 2024-10-15:
No I am just asking to add this so that a small but thick wipe line can be made near the beginning of the print.  Klipper has it and so could Cura if it had those variables

https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging