Ultimaker/Cura #19749: Unusable with Narrator or NVDA screen readers

ClovernookQuentin, 2024-10-14:
### Cura Version

5.7.1 and 5.8.1

### Operating System

Windows 11

### Printer

All

### Reproduction steps

1. Start Narrator
2. Open new install of Ultimaker Cura
3. Attempt to add printer without looking at screen using tab key to cycle through

### Actual results

Unable to set up Cura for use due to the add Ultimaker Printer and add Non-Ultimaker Printer buttons not actionable through keyboard inputs.

Notes from Blind User:
Slicer does not speak when loaded.
Many buttons are unlabeled.
Cannot properly tab through the application.

### Expected results

Ability to use Software without sight.

### Add your .zip and screenshots here ⬇️

[image]

Image of the screen where the buttons are unactionable and unable to be selected making the software unusable.

GregValiant, 2024-10-14:
Thanks for this.
I don't know that it is a bug, or something that was just never implemented.  This issue might be changed to a feature request.

The controls that Cura uses are "QT Controls" which are compatible with multiple operating systems.  It appears that the tab order is editable, but I don't have any experience with that.

Someone from the Cura team (@nallath ?) will take a look.

ClovernookQuentin, 2025-04-24:
Still needs fixed. Currently blind users cannot use our Ultimaker printers.