Ultimaker/Cura #19750: can't use full height of the machine

Rick-TRIGGER, 2024-10-14:
### Cura Version

5.8.1

### Operating System

Mac OS sonoma

### Printer

Artillery sw-x2

### Reproduction steps

just want to print a 300mm part up but the built plat show me only a 200 mm... 
the parameter in the machine was 300x300x400 

### Actual results

can print a part longer tahn 200 mm on the Z axis

### Expected results

cant print

### Add your .zip and screenshots here ⬇️

[image]

[image]

Rick-TRIGGER, 2024-10-14:
[image]

Rick-TRIGGER, 2024-10-14:
the last version that work was cura 5.6 on macx64

GregValiant, 2024-10-16:
Thanks for the report.
Please use the File | Save Project command and then zip that new 3mf file and post the zip folder here.  We need something to investigate.

Rick-TRIGGER, 2024-10-17:
I find that the probleme apear since 5.7.5.6 was okEric Trudel450 521-9330 ***@***.*** 16 oct.

closed 2024-11-01 by github-actions[bot]