Ultimaker/Cura #19759: Updates / Marketplace+Materials

jacc190676, 2024-10-16:
### Is your feature request related to a problem?

Hi

It's nice to see that an Update (or something) is availlable (red number), but:
sorry, the list of addons is endless. everytime i start cura, i have a look what this (sinse days same) update/setting change should be. i don't find it ! i simply not find it (UPDATE, NOW after the 10 try i found it, but sorry, no...., that needs to get better!). Would it be possible to mark updates colored or even better put it at start of the list of all addins? Or then an own button "Updates" wich only lists the updates?

Thank you

### Describe the solution you'd like

updates get found mutch easyer

### Describe alternatives you've considered

tried all i can (scrolled many times up and down, restarted, checked again. no solution found (Update, found a solution after don't know how much search tries, THIS can not be a final solution to search everytme so intensive)

### Affected users and/or printers

i think all

### Additional information & file uploads

[image]

LilBub, 2024-10-16:
Yes to this...and list them at the top with check box to "Update selected"

PartySausage, 2024-10-16:
I agree it would be good if the list was populated with updates at the top of the list.
The latest update that came up for me was for a Ultimaker PPS-CF material profile update. It didn't always show as an update  pending when I went to the material manager list & sometimes the Marketplace would close, whilst I was scrolling through the list. 
When it eventually did show the update, I selected for it to update & restarted Cura I just get an error message & the package doesn't update so doesn't clear the update flag. I raised it as a separate bug report
[(https://github.com/Ultimaker/Cura/issues/19722) ](https://github.com/Ultimaker/Cura/issues/19722)

mevanloon, 2024-10-27:
Yeah I think this can be tagged 'bug'. That red badge being there consistently with no way of (obviously) removing it, that's not intended behavior.