Ultimaker/Cura #19765: Missing the 'CuraEngine Tiled Infill Generation' Plugin

Bedslinger, 2024-10-16:
### Cura Version

5.8.1

### Operating System

Windows 11

### Printer

Creality Ender 3/Ender-3 v2

### Reproduction steps

1. Clicked on 'Marketplace'
2. Searched for 'CuraEngine Tiled'

### Actual results

The desired plugin (displayed in this video: https://www.youtube.com/watch?v=IOpm3cJbdHQ ) is not shown

### Expected results

The 'CuraEngine Tiled Infill Generation' plugin is shown as one of the plugins that comes up when its name is searched via the Marketplace

### Add your .zip and screenshots here ⬇️

[image]
[image]

GregValiant, 2024-10-16:
Thanks for the report.
It may have been removed for some reason.
@HellAholic ?

GregValiant, 2024-10-17:
@Bedslinger,
There you have it.  I'll remove the bug label and close this as the removal was intentional.

If you have a previous version of Cura that included the plugin, and if it wasn't causing you any trouble (it did not for me) then you can copy the folder from your configuration folder ("Help | Show Configuration Folder") and the "plugins" sub-folder to the same folder in your new version.
[image]

closed 2024-10-17 by GregValiant