Ultimaker/Cura #19772: Slicing failed on all STL files

SaixLionhart, 2024-10-17:
### Cura Version

5.8.1

### Operating System

Windows 10

### Printer

Anycubic Kobra Max

### Name abnormal settings

does not slice with default profiles either

### Describe model location

model sometimes slices if placed 50mm above buildplate. others need to be 100mm above buildplate.

### Describe your model

I have sliced other models successfully months ago. Meshtools says the model is watertight.

### Add your .zip here ⬇️

I have uninstalled/reinstalled my graphics driver. Uninstalled all versions of Cura, cleared cache from the Appdata folders, and installed multiple versions of Cura (5.6 to 5.8). I have updated my OS to Windows 11 and when it did not work still I downgraded by to Windows 10. This issue also happens on PrusaSlicer (application will just force close itself), however it does not happen in LycheeSlicer.
[AKMAX_Articulated crystal dragon.zip](https://github.com/user-attachments/files/17414984/AKMAX_Articulated.crystal.dragon.zip)

GregValiant, 2024-10-17:
Thanks for the report.
I've been doing triage here for a few years now and out of all the models I've investigated - the original "Crystal Dragon" was the worst with over 30,000 errors in it.
This version is fine and I didn't find any errors at all.  
At any rate, I didn't have any trouble slicing this with either 5.8.0 or with 5.8.1.

PrusaSlicer and Cura don't have much in common codewise.  It sounds like your system still is getting heartburn from the softwares (rather than some common bug to both slicers which is extremely unlikely).
What processor are you running and what video system does the computer have?
Are you running multiple monitors and if so, are they the same brand?

Use the "Help | Show Configuration Folder" command to locate the Cura.log file.  Set up the slice again and immediately after slicing fails - zip the log file and post it here.  Maybe it will have a clue inside since I'm unable to duplicate the problem on my Win10Pro system.

SaixLionhart, 2024-10-19:
Hi Greg. Thank you for the support! To answer your questions:

Processor: Intel Core i9-14900KS Raptor Lake 3.2GHz 24 Core LGA 1700
Video: MSI Nvidia GeForce RTX 4080 Super Gaming X Slim 16GB
Monitors: 2 Acer monitors (32" ED320QR S and 23" G236HL) and 1 Asus monitor (27" VP278)

Attached is the Cura.log file as requested.
[cura.zip](https://github.com/user-attachments/files/17447607/cura.zip)

I can't tell you exactly when this started happening, but it was maybe a month or two ago. I built a new PC with the above components back in May/June this year and my previous system was running an i7-12000 processor with a GeForce 1070 and I did not run into this issue.

Please let me know if you need any other info from me!