Ultimaker/Cura #19774: not waiting for Hotend Temperature

subcamera, 2024-10-18:
### Cura Version

5.8.0-beta.1

### Operating System

Windows 10

### Printer

Ultimaker 2+

### Reproduction steps

creating gCode wirg Cura
The first block sets initialises tempereture of Heatbed an hotend.
BUT printer ist nor waiting for reachin hotend-Temperature
Instead of M104 gcode M109 should be used.

### Actual results

After changing manualy first "M104 Sxxx" to "M109 Sxxx" everything worked fine.

### Expected results

Please correct Cura gCode header.
THX

### Add your .zip and screenshots here ⬇️

no files needed

Asterchades, 2024-10-19:
This is a duplicate of #19398.

You'll find more detail about it there, but the short of it appears to be that setting the Gcode Flavor to Marlin, instead of Ultimaker, is what causes this behaviour. Restoring this setting should make it work. If you absolutely need to use Marlin instead (such as for OctoPrint) you can work around this by either using the *Printer Settings* plugin to enable the *Wait for Nozzle Heatup* option or by manually setting the Start Gcode to include an M109.

GregValiant, 2024-10-23:
@subcamera I'm going to go ahead and close this in favor of the previous report (noted by Asterchades).  I hope you understand.

closed 2024-10-23 by GregValiant