Ultimaker/Cura #19777: Slicing hang on 30%

Kilinich, 2024-10-18:
### Cura Version

5.8.1

### Operating System

Windows 11

### Printer

Voron2 350

### Name abnormal settings

normal settings

### Describe model location

Not sliced anywhere

### Describe your model

It's a plate with a lot of holes

### Add your .zip here ⬇️

[Door.zip](https://github.com/user-attachments/files/17440765/Door.zip)
Slice starting and hang on 30%. No cpu load, no progress.

GregValiant, 2024-10-19:
Thanks for the report.
The slicing bar did hang up on around 25% to 30% for a long time (in fact it was taking so long that I went to bed).  From my log file I see "Slicing took 2805.5770180225372 seconds" which is ~47 minutes.  I would suggest turning off "Group Outer Walls" for this model.
It looks like this when sliced:
[image]

When I changed a couple of settings that my experience has shown can effect slice time (Wall Count = 3, Top Surface Layers = 0, Infill Pattern = Grid, Infill Density = 10%, Combing = All, Group Outer Walls = False) the slicing bar hung at 50% and slicing time was Slicing took 2025 seconds which is an improvement of about 13 minutes.  Still a long slice.
Work was done to 5.7 (I think) and slicing time was improved dramatically.  In the case of this model, it might be that there are just so many features per layer that it's still a very long slice.

If you can't get it to finish slicing, then I would consider that a bug.  If it simply takes a long time (as it did for me) then it is likely this model is the problem as I haven't noticed problems with other models.

Kilinich, 2024-10-19:
Well, yes, disabling groups makes it possible to slice at a "reasonable"
time of about 30 minutes. But then I'm trying to look at first layer
movement (touch bottom progress bar control) - Cura crashed:

[image: image.png]

GregValiant, 2024-10-19:
The crash when trying to show the first layer preview is a different problem and is likely either a video driver compatibility problem, or a problem with the way Cura tries to display the preview.  The initial problem here was the Slicing error.
I can't duplicate either problem on my system using either 5.8.0 or 5.8.1.  I'll leave the "Slicing Error" label on it though and the Cura team will take a look.