Ultimaker/Cura #19780: trying to slice a duodecahedron and slicing reliably fails on the stl.

interrupt80, 2024-10-20:
### Cura Version

5.7.1

### Operating System

Windows 10 Pro/ NVIDIA GeForce GT 730

### Printer

Creality ender-3 S1

### Name abnormal settings

standard profile(TPU) work, moving to gyroid and 5 % infill works,  wall to .4 from .8 causes crash

### Describe model location

center

### Describe your model

No

### Add your .zip here ⬇️

[CE3S1_dodecahedron_crash.zip](https://github.com/user-attachments/files/17450494/CE3S1_dodecahedron_crash.zip)
original modeL: https://www.thingiverse.com/thing:198

GregValiant, 2024-10-21:
Thanks for the report.
I can't duplicate the problem.  The project slices for me on my Win10Pro laptop.

Locate your Configuration Folder ("Help | Show Configuration Folder") and the cura.log file.
In Cura, set up the slice again and immediately after it fails - zip the **cural.log** file and post it here.  (It must be posted directly and not via email.)

[image]

interrupt80, 2024-10-21:
[cura.zip](https://github.com/user-attachments/files/17452997/cura.zip)

interrupt80, 2024-10-21:
I have a path forward, I just like getting bugs to go away.

GregValiant, 2024-10-21:
From your log file.
_"Backend exited abnormally with return code 3221226505!"_

There were several bug reports that mentioned that code.
@HellAholic and the Cura team have attempted to fix this in newer releases.  You can try running 5.7.1 "As Administrator" or maybe upgrading to Cura 5.8.x.

I'll mark this as a duplicate and close it as it appears this was fixed.  If the problem still exists in 5.8.x then let us know and this can be revisited.

GregValiant, 2024-10-21:
Since this occurred with 5.7.1 is this an ongoing issue?  If it is then I'll re-open this ticket.

interrupt80, 2024-10-22:
> Can't say for sure, since I can't reproduce it either, so we have no way of checking if it is resolved or not. Main part was getting the model, I'll add it to the automated slicing test so hopefully if there is a crash or regression on the model it will catch it. TLDR: It's fine!

5.8.1 resolves it for me