Ultimaker/Cura #19781: Cura GCODE misses part of the object

kirevdokimov, 2024-10-20:
### Cura Version

5.8.1

### Operating System

Windows 11

### Printer

Creality V3 SE

### Reproduction steps

1. Open The Project File
2. Click Slice

### Actual results

Invalid g-code after layer 44. See Details below.

### Expected results

Valid and complete G-Code. 
The model is printed as it is shown in the preview.

### Add your .zip and screenshots here ⬇️

- A few layers of the 3d model are missing. GCode inspection with https://ncviewer.com/ shows issues:
[image]

- Meanwhile, the preview is totally fine
[image]

- But the print is not complete as well
[image]

- The project file archived as a zip to attach to the issue
[550Mount.zip](https://github.com/user-attachments/files/17452226/550Mount.zip)

- Output g-code:
[550Mount.gcode.txt](https://github.com/user-attachments/files/17452233/550Mount.gcode.txt)

- G-Code is not ok for sure
[image]

kirevdokimov, 2024-10-20:
I turned the model and then turned back to the initial orientation and sliced again. This time no issue and the g-code looks valid
[image]

But I have many issues like this during the day

GregValiant, 2024-10-20:
Thanks for the report.
Did you save the Gcode to your hard drive or directly to an SD card?  There are occasions when a file is written (by the operating system) to a bad memory sector on an SD card and the result is gibberish.  It kinda-sorta ends up looking like that.  Cura passes the gcode text file to the operating system to save to disc so that garbage at the end wasn't Cura.  
If you are sending the Gcode to the printer some other way (Octoprint, Creality  Sonic Pad, etc.) let us know.

I'll leave the bug label on this for now, but I think the issue is somewhere else rather than Cura.  No one else has mentioned this behavior (other than corrupt memory cards) and if it was a problem with Cura people would be lining up to complain.

kirevdokimov, 2024-10-21:
Thank you for your quick and detailed response @GregValiant!

You are correct, I am saving the g-code directly to my SD card which is unbranded and was included with the printer. Thinking from the point of view you gave me, I assume my tendency to quickly eject the card right after it is written might contribute to the issue as well.

I will try different SD cards as well as saving to the disk first to know for sure. I think the issue may be closed :)

GregValiant, 2024-10-21:
You can try reformatting the card (to FAT).
I'm using the same 16gb card that came with my Ender.  It's been reformatted at least 10 times and it's still working.

Some of the early Creality printers had a hard limit to 16gb cards.  I think the newer models have upped that to 32gb.  If the card is too large the printer may not read it unless it is partitioned (maybe!!).
Yes, improperly ejecting the card can cause issues if files are opened in an application (even in File Explorer).