Ultimaker/Cura #19788: Cura will not open when multiple screens are active in Windows 11

DaveTPilot, 2024-10-22:
### Cura Version

5.8.1

### Operating System

Windows 11

### Printer

Creality CR6 SE

### Reproduction steps

I have "Extend These Displays" active in Windows 11. 
I have 2 monitors. 
Tried to start Cura.

### Actual results

The splash screen appears as normal.
The workspace then attempts to open on the second screen.
Cura locks up and will not start. 
Cura will not respond. 

### Expected results

Cura has always worked in this configuration before.

### Add your .zip and screenshots here ⬇️

I have no other information to share.

Asterchades, 2024-10-23:
This shows up periodically and is something that never has one single answer for all users.

The issue stems from an incompatibility between the Qt framework (which Cura uses for display) and *something* else on the user's system. I emphasise "something" because there genuinely isn't one other piece of software that can interfere - and not all of them are immediately obvious that they might be having an effect. For example, the _Nahimic Service_ is known to interfere, and it's an audio enhancement program! I also know that the *FancyZones* feature of the modern *PowerToys* suite can cause problem, but I never experienced it causing a load failure (for a little while it did make the *Sidebar GUI* plugin run terribly).

To that end, there is something else you've either installed or updated which has resulted in Cura no longer working properly on more than one screen. If you can identify this and either disable it or roll it back that should let you run Cura again without issue. If you're able to find such a program I would encourage you to report back with the name and version (if possible) so it can be both tracked for posterity and also investigated.

Alternatively, and I don't know how successful this is, you could try starting with a fresh config file. Cura stores its last window position by default and that has been known to interfere on some systems (but not all). You don't need to remove/rename everything - just the _cura.cfg_ file you'll find in _%appdata%\cura\5.8_ (this won't remove any printers, but will reset things like the current display theme, setting visibility, and some plugins like OctoPrint keys).  If this works, make sure to uncheck the _Restore window position on start_ option in the preferences.

DaveTPilot, 2024-10-23:
Thank you for the honest and detailed response! It is so refreshing to get
useful information as opposed to the standard, “Have you tried rebooting
your computer?”

Before receiving this email, I disabled the second monitor and was able to
start Cura successfully. I then unchecked the
*Restore window position on start (*as you suggested below) and reenabled
the second screen. Cura worked perfectly unless I moved the window to the
second screen, then it locked up again. CTL+ALT+DEL, End Task.

Regarding programs/apps/updates, you are correct. I had updated several
drivers and installed Armoury Crate for my ASUS STRIX. It was the first
thing I uninstalled when Cura stopped working properly but it did not help.
However,  after reading your response, I deleted the cura.cfg file. Cura
now works on both screens again! I assume that Armoury Crate was the
offending app but required the deletion of the cura.cfg file after
uninstalling the app.

I hope this information is useful.

Thank you again.

pontuslaestadius, 2024-10-25:
Experienced the same issue the past two weeks, haven't been able to launch the program. Reinstall etc. did not work but removing the cura.cfg did the trick, thanks @Asterchades. Further testing showed that just removing the following from the [general] section in the cura.cfg also worked:

`
window_state = 2

window_left = 2251
`

Asterchades, 2024-10-25:
Appreciate the feedback. Will keep in mind the specific config lines as well, since it's always less hassle to reset as few settings as possible.

I'll be closing this one up for now as it seems that this has been resolved.

closed 2024-10-25 by Asterchades