Ultimaker/Cura #19801: 5.8.1 blobbing on prints in identical locations

Sprlnkle, 2024-10-24:
### Cura Version

5.8.1

### Operating System

Windows 10

### Printer

Ender 3 v3 SE

### Reproduction steps

Setup: ender 3 v3 se with moonsail + klipper on rpi4 connected via usb. When slicing a mesh with a high mesh accuracy (0.15 or 0.25) blobs occur on the print in identical locations every print. I am positive this is not the z seam or bubbling from filament because the dots are in the same location every single time and the z seam is on the back.

### Actual results

Blobs on the bow of the benchy or any curved surface with a high mesh accuracy (which leads me to believe it is a buffering issue).
After attempting the same model on cura 5.2.0 the blobs did not occur and the print was perfect. This can be prevented by decreasing the mesh accuracy to a degree where print quality is significantly impacted and I initially thought it was a flaw with my setup but mesh accuracy of 0.15mm is achievable in older versions which leads me to believe it is a software bug.

### Expected results

The benchy printed with cura 5.8.1 should not have blobs on curved surfaces. This can be prevented by decreasing the mesh accuracy to a degree where print quality is significantly impacted and I initially thought it was a flaw with my setup but mesh accuracy of 0.15mm is easily achievable in older versions which leads me to believe it is a software bug.

### Add your .zip and screenshots here ⬇️

[image] (sliced benchy in 5.8.1)
[image] (resulting benchy note the zits are always in the same location)
[image] (benchy sliced in 5.2.0)
[image] (resulting benchy bow sliced in 5.2.0

GregValiant, 2024-10-25:
Thanks for the report.
The actual resolution of the model is dependent on the resolution that the STL (or 3mf, OBJ, etc.) file was created with.  Nothing can be done by a slicer to change that.

The setting for Maximum Resolution is a filter to catch extremely short line segments that can result from the slicing of arcs and curves.  A lower number allows shorter line segments through.  Differences in the slicing algorithms (and changes are always being made to Cura) can result in a particular printer requiring a different setting for the Maximum Resolution.
A MR setting that is too low can result in both a bloated gcode file, and cause blobbing problems if the input buffer of the printer cannot keep up with the printer/processor.  The blobs are caused by oozing as the print head hesitates while it waits for the next gcode line to be processed.  

The base "Maximum Resolution" of Cura's "fdmprinter.def.json" file is 0.50.  Creality addresses that in their "creality_base.def.json" file in line 214 with:
_"meshfix_maximum_resolution": { "value": "0.4" }_
No further change to the Maximum Resolution is made in any Creality _machine specific_ definition file.

In earlier Cura versions, Creality had made an attempt to set the Maximum Resolution to 0.25 but that caused exactly what you see in your print.  It was just too low for the printer to handle.  I run my Ender 3 Pro at 0.4 MR and it's fine.

Here are two models.  They are actually the same model but I exported one from CAD at a high resolution and then exported it again at a low resolution.
[image]
There is no way for a slicer to turn that low resolution model into a high resolution model.  The setting for "Maximum Resolution" can have no effect on that because the resolution is built into the model file.  What a reasonable "Maximum Resolution" can do is allow the hi-res model to print without hesitating and leaving blobs.

I'm inclined to remove the bug label from this since it appears you simply have Maximum Resolution set too low for the combination of Cura 5.8.1 and your printer.

Sprlnkle, 2024-10-25:
> Thanks for the report. The actual resolution of the model is dependent on the resolution that the STL (or 3mf, OBJ, etc.) file was created with. Nothing can be done by a slicer to change that.
> 
> The setting for Maximum Resolution is a filter to catch extremely short line segments that can result from the slicing of arcs and curves. A lower number allows shorter line segments through. Differences in the slicing algorithms (and changes are always being made to Cura) can result in a particular printer requiring a different setting for the Maximum Resolution. A MR setting that is too low can result in both a bloated gcode file, and cause blobbing problems if the input buffer of the printer cannot keep up with the printer/processor. The blobs are caused by oozing as the print head hesitates while it waits for the next gcode line to be processed OR it isn't hesitating but is printing a series of extrusions that are each 1 micron long. Either way the printhead is effectively sitting in one spot for a split second, and leaving the blob.
> 
> The base "Maximum Resolution" of Cura's "fdmprinter.def.json" file is 0.50. Creality addresses that in their "creality_base.def.json" file in line 214 with: _"meshfix_maximum_resolution": { "value": "0.4" }_ No further change to the Maximum Resolution is made in any Creality _machine specific_ definition file.
> 
> In earlier Cura versions, Creality had made an attempt to set the Maximum Resolution to 0.25 but that caused exactly what you see in your print. It was just too low for the printer to handle. I run my Ender 3 Pro at 0.4 MR and it's fine.
> 
> Here are two models. They are actually the same model but I exported one from CAD at a high resolution and then exported it again at a low resolution. ![image](https://private-user-images.githubusercontent.com/64202104/379974927-b0148340-07e3-4307-b429-0939f0b96008.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk4MTk1MDAsIm5iZiI6MTcyOTgxOTIwMCwicGF0aCI6Ii82NDIwMjEwNC8zNzk5NzQ5MjctYjAxNDgzNDAtMDdlMy00MzA3LWI0MjktMDkzOWYwYjk2MDA4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI1VDAxMjAwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY3MTQ0MDM2NDNhNzYyMTYxY2IyODYwZThmMDg0ZjcxZmFjZTAzZDcxMTRiOTI1NDhmNTYxZTgzNzVhOWFjYWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.NAKwOYLZjeG9nuKK8tXObPWoWMbVGBwq22bPKSZpfSQ) There is no way for a slicer to turn that low resolution model into a high resolution model. The setting for "Maximum Resolution" can have no effect on that because the resolution is built into the model file. What a reasonable "Maximum Resolution" can do is allow the hi-res model to print without hesitating and leaving blobs.
> 
> I'm inclined to remove the bug label from this since it appears you simply have Maximum Resolution set too low for the combination of Cura 5.8.1 and your printer.

I understand to some extent that the printer is being overwhelmed by the gcode being too big or too fast. however even on a setting of .5 or higher depending on how fast the blobs are still apparent due to the printer stalling out. At the point where the blobbing stops at a reasonable print speed the model has been physically mangled due to the low resolution. Not just that but an older version of cura had the capability to produce a benchy with the resolution at 0.15 with no buffering issues whatsoever. I feel this should be classified as a bug because any person who wants to make high resolution models like figurines or tight tolerance models can no longer make those models and must either use an older version of cura and be barred from utilizing new features or forced to use another slicing software as it is, at least on my setup, virtually impossible to create a high quality print as things stand if print speed is higher than 20ms whereas in cura 5.2.0 I can make the same high resolution model at speeds in excess of 100ms.

Sprlnkle, 2024-10-26:
Further evidence that something must be wrong even after reducing the resolution from .15 (which is still printable in older versions) down to .5 at 60mm/s which should be easily achievable the result is still the same with major blobbing around rounded corners whereas the same settings on older versions yield perfect benchy prints. Something else must be wrong with cura that was changed in some update. Between 5.2.0 and most recent. I know at one point there was a similar issue that was supposedly patched from research but I think there may still be some issue as I have tried tweaking every setting and nothing has fixed the issue other than running an older version of cura or a resolution of >1mm which obviously results in an unacceptable print quality. (Thank you for taking the time to hear me out I've been testing and trying to fix this for at least 5 months)
[image]

Sprlnkle, 2024-10-26:
Perhaps it is a printing over usb specific bug then? For a long time I thought it was something to do with my personal setup but I have tried everything imaginable. I think it has something to do with printing over usb + recent cura. I might try downloading many versions of cura to find where the problem starts as that may help us pinpoint what is going on. I know not a lot of people ever end up setting up anything like klipper so there may not be a lot of people to experience the issue or perhaps they presume their setup is the issue similar to what I did for months so nobody makes a report. I will keep testing to try and determine if it is just my printer or something else.

raska-vilem, 2024-10-28:
Hello guys,

I think I have the exact same problem. I opened [this thread](https://community.ultimaker.com/topic/46295-zits-on-rounded-walls/) on the forum about it.

My current status is:
1) The zits/blobs (or how to call that) are created by Cura, it can be seen in gcode here:
[image]

2) My 3D model is also in high resolution (I‘ve created that in OpenSCAD with `$fn=300`)

- OS: MacOS 15
- Cura: 5.9.0-beta.1
- Printer: Crelity Ender 5 + Klipper and many upgrades

Sprlnkle, 2024-10-30:
> I tried Cura 5.2.2 as @Sprlnkle mentioned. The resulting surface was great, but the z-seam alignment was worse. I tried other slicers (PrusaSlicer, SuperSlicer, LycheeSlicer), but the prints were significantly worse overall.

Do you have a finished print with that gcode? After observing my gcode I didn't see the same kind of blemish on the print. My gcode looks fine but my print has blobs (cura 5.8.2). . Curiously 5.2 works for us both so perhaps a good starting point for us would be to determine where the problem starts. I also think we should discover whether or not we are experiencing the same issue or a different issue before trying much else to avoid muddying the water.

GregValiant, 2024-10-31:
The bug label is still on this and the Cura team will take a look, but I don't get the problem in my prints.

The larger model has a slight change in color where I had the fan go to 100%.  The radius is 40mm.
The benchy is in PETG and looks like a benchy.  It certainly doesn't have the issues I see in the prints up above.  They were both sliced with 5.8.0 and printed on an Ender 3 Pro.
[image]

raska-vilem, 2024-10-31:
I was trying to reproduce that issue on some generic 3D model, but without success. I am attaching a zip file with `.stl`, `.3mf`, `.scad` and `.gcode` => [blobing-test.zip](https://github.com/user-attachments/files/17587786/blobing-test.zip). I'll try to investigate further.

Sprlnkle, 2024-11-08:
> The bug label is still on this and the Cura team will take a look, but I don't get the problem in my prints.
> 
> The larger model has a slight change in color where I had the fan go to 100%. The radius is 40mm. The benchy is in PETG and looks like a benchy. It certainly doesn't have the issues I see in the prints up above. They were both sliced with 5.8.0 and printed on an Ender 3 Pro. ![DSCN3344](https://private-user-images.githubusercontent.com/64202104/381786437-ada22165-6281-4354-8bf5-4d8c18d383e7.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzExMDE4NTIsIm5iZiI6MTczMTEwMTU1MiwicGF0aCI6Ii82NDIwMjEwNC8zODE3ODY0MzctYWRhMjIxNjUtNjI4MS00MzU0LThiZjUtNGQ4YzE4ZDM4M2U3LkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTA4VDIxMzIzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQxY2MwZjIwNWViZDczNmMxNjJkMmIxYzEwNDVhYjAzZDY0ZDg0Y2RlYTUxYThhMDFmM2I4MjYzOTQ0NWE1NzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ZTyiyFT2c3UJSrxCe4boFJViJUPNTg2Uuy7kZ0eBES0)

Did you try printing over usb?

raska-vilem, 2024-11-08:
I was printing only over USB from RPi 3b+

Sprlnkle, 2025-01-20:
I have found a solution to this issue. I don't know how or why this is but having the Ender 3 v3 SE preset selected in Cura while printing using a raspberry pi with klipper and octoprint or moonsail ect does not work. I don't understand why this is but to anyone having a similar issue to this try just setting up a custom FFF printer in cura rather than using your printers preset. (I don't know if Cura formats the Gcode any differently for different printers, to my knowledge a gcode was a gcode and as long as the correct start/stop ect was in place for the print to work at all it shouldn't have any strange artifact like this so I'm still just as confused but the problem is solved)