Ultimaker/Cura #19804: [5.9.0-beta.1] - Scarf seam disappointing performance

Hello1024, 2024-10-25:
### Cura Version

5.9.0-beta.1

### Operating System

Linux

### Printer

Anycubic Kobra Max with klipper

### Reproduction steps

Enable Scarf Seams.

Observe print results.

### Actual results

Regular seam with wipe:
[image]

Scarf seam:
[image]

This is a sphere with "sharpest vertex", so the seam location jumps about a bit.   Note how there is a distinct blob at the seam.

For the regular seam, there is a small blob at the start/end of the print (the right hand blob), and another blob at the end of the wipe.

For the scarf seam, we see a big blob at the start and end of the scarf, but also smaller blobs at every vertex along the scarf (presumably because there are brief stops/slowdowns there to change the z height)

Do others have better results here?

### Expected results

No visible seam.

### Add your .zip and screenshots here ⬇️

[sphere_r-24mm (1).zip](https://github.com/user-attachments/files/17520819/sphere_r-24mm.1.zip)

GregValiant, 2024-10-27:
Thanks for the report.
After seeing a couple of reports I did some experiments.
This is a spliced Gcode.  Regular Zseam at the bottom, scarf Zseam at 0.5 in the middle, and scarf seam at 1.0 at the top.
It will always be tough to totally hide the zseam because it is different than the adjacent extrusions and reflects the light differently.
The 1.0 setting of the top portion resulted in a nearly flat seam.  I like that as it would be good for printing things like threads.  I don't like the Printing Time hit that the Scarf Seam causes (about +30% on my experiments).
[image]

After reading the settings for the model that UM posted on Thingiverse I tried Scarf seam at 15.0 with a 2 wall print.
I'm not impressed with this one.  I sliced the model with 2 walls and no infill.  The "inside" seam was very noticeable.
[image]

Hello1024, 2024-10-27:
How many walls is this with?

There seem to be quite different results with 1 vs 2+ walls, since in one
case any blob can occur in an inwards direction.

Then there are two or more walls, the print order of the walls seems to
have quite a big difference.  outside to inside is most easy to reason
about, but on most printers has a negative impact on overhang angle.
 Inside to outside, it really is important to have the scarf seal on inner
layers too, since any plastic blobs on an inner seam will propagate
slightly to the outer.

re:  print time.    I think it will be possible to eliminate the print time
hit of scarf seams by adjusting the algorithm to reduce 90 degree corners
at the start and end of travel moves when printing scarf seams - since
every 90 degree corner involves the print head slowing down a lot to go
round the corner.    This also has benefits of making blob-free printing
more likely, since every time the print head slows down whilst printing, in
the ideal case volumetric flow rate goes down proportionally, but in
reality this depends on temperature, viscosity, pressure advance config
etc.    These settings will never be perfect, and models will never
perfectly reflect reality - so it's better to avoid this situation entirely
by avoiding cases where the print head needs to slow down whilst printing
by avoiding sharp corners of travel moves.

Hello1024, 2024-10-27:
Also see here:   https://github.com/Ultimaker/Cura/issues/19805

I have tried a different approach and achieved invisible seams - if you are up for compiling code you could try it.

(it's far from finished - the code is ugly and there is no setting to control it, but it does result in no visible seam, even with inside-to-outside wall printing order)

josh9465, 2025-01-07:
[image]
I get a line blob if I use scarf