Ultimaker/Cura #19805: Better seam approach

Hello1024, 2024-10-25:
### Is your feature request related to a problem?

Better approach to seams being visible.

### Describe the solution you'd like

Make the seam be gradual by starting the extrusion from inside the shape, gradually increasing flow rate and moving outwards, doing one complete lap of the object, and then reducing the flow and moving inwards again to finish.

Like this:
[image]

### Describe alternatives you've considered

Scarf seams didn't work well for me.

### Affected users and/or printers

Everyone.

### Additional information & file uploads

I have implemented this (very roughly!)

See here:    https://github.com/Hello1024/CuraEngine/tree/Hello1024-better-seam

It works well:

my implementation:
[image]

regular seam:
[image]

scarf seam:
[image]

I'm not sure I'll have time to bring this code up to quality for contributing to the project, but wanted to document that this approach works, and as well as getting rid of the seam, also has a ~5% print time saving due to reduced travel moves and the travel moves having no 90 degree edges which regular perimeters have.

wawanbreton, 2024-10-25:
Hi @Hello1024,

Thanks for the report, this is extremely interesting indeed :slightly_smiling_face: 
We did a try for a different approach move a few months ago, but parked it because we had many edge-cases and did not give that good results. But your approach looks very much like a scarf seam, but in the XY plan instead of Z. So it could actually be added to our actual scarf seam algorithm very easily (quite like you did actually). I'll probably give it a try.

However, I think there may be a patent on this technique, in which case we would not be allowed to implement it. I may be mixing it up with something else though, I'll check with the team on Monday.

Hello1024, 2024-10-25:
@wawanbreton please look in detail at [this diff](https://github.com/Ultimaker/CuraEngine/commit/99ffc65d42b2bfa7044fdd18dc1b07f4412bf8a5).   I believe that resolves your concerns, but if it doesn't it's probably a 5 min [offline](http://omattos.com/2013/08/03/My_phone_number.html) topic.

wawanbreton, 2024-10-25:
> I believe that resolves your concerns

I fail to see how :thinking: I would have made a setting of this distance tbh, since this is typically the kind of thing that people will want to customize.

Hello1024, 2024-10-25:
That specific number is critical to the patent you referenced.  All numbers 2 or below would violate the patent.

In my experience, ideal print quality is achieved at 2, but you get decent results all the way between 1 and 4.    2.1 seems like the best compromise, and obviously the 10 micrometer difference wont affect print quality.

Hello1024, 2024-11-01:
By the way, this change also seems to really help the stability of tree supports.    Tree branches don't break as easily because they don't have a weak spot at the seam.

Have now printed quite  a lot with it, and the approach seems robust, despite the code being a giant mess.    Pondering how to tidy it up - any thoughts @wawanbreton ?

wawanbreton, 2024-11-15:
Hi @Hello1024, we are currently in contact with some people in the company to try and figure our whether implementing this would indeed be possible. We will keep you posted as soon as we have more reliable information !

Hello1024, 2024-11-15:
Great.    Note that the image in my original post has the value in question
as 1, whereas I would recommend 2 or more to make sure the 'road'
(extrusion line) falls within the 'perimeter' (outer wall).   I will make
another image with the value as 2 to better explain the concept...

Hello1024, 2024-11-15:
This is with the parameter set to above 2.   Notice how the wall starts within the outer perimeter.

[image]

Video so one can see it better:
https://youtu.be/owhnTYUNPGc

Hello1024, 2025-04-03:
Yes.   Specifically "wherein at least one of the start point and the stop point is located within the interior region of the layer".    Values below 2 would involve the start and stop points being within the interior of the layer, whilst values above 2 are within the perimeter of the layer.