Ultimaker/Cura #19810: [5.9.0-beta.1] Print Head Drops to Z=0 When Coasting is Activated

cytron369, 2024-10-26:
### Cura Version

5.9.0-beta.1

### Operating System

Linux

### Printer

Creality Ender 3 SE

### Reproduction steps

1. Open Ultimaker Cura.
2. Load any 3D model for printing.
3. Enable the Coasting feature in the print settings.
4. Start the print and observe the behavior of the print head when reaching a seam or corner.
5. Note that the print head drops to Z=0 instead of maintaining the correct print height.

### Actual results

The print head drops to Z=0 when reaches seam and corners in any model.

### Expected results

he print head should maintain the correct print height and not drop to Z=0 during seams or corners, even with Coasting enabled.

### Add your .zip and screenshots here ⬇️

[test.zip](https://github.com/user-attachments/files/17528341/test.zip)

cytron369, 2024-10-26:
In the gcode (in test.zip) you can see Z0.00 belong the file (in seam and corners each layer).

odenibsb, 2024-10-29:
The same happens here with a kossel mini diy.

alfredogluque, 2024-10-31:
Hello everybody!   I can confirm this bug, it happened to me also. The 'Z0' gcodes don't appear if I disable Coasting. Thanks Cytron for pinpointing the culprit parameter.
Cheers!