Ultimaker/Cura #19812: Trous non percés, qui sont présents dans le .stl

MichelBrai, 2024-10-26:
### Cura Version

5.7.0

### Operating System

Windows 10

### Printer

Artillery genius

### Reproduction steps

1 j'ai créé mon fichier .stl qui a trois trous et je l'ai importé dans Cura
2 Mon fichier apparaît dans Cura, il a bien ses trois trous
3 J'ai fait le découpage, puis sans le vérifier, je l'ai imprimé en 3D

### Actual results

Les trois trous sont bouchés.
J'ai refait l'expérience, les trois trous sont bouchés après le découpage.
J'ai essayé avec Cura 5.9.0 beta 1, j'ai le même problème.
J'ai essayé avec un concurrent, Creality slicer, les trois trous sont apparents après le découpage, tout est normal.

### Expected results

Les trous devraient être faits.
[3D Objects.zip](https://github.com/user-attachments/files/17530635/3D.Objects.zip)

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-10-27:
Thanks for the report.
From your Gcode file line 84113:
**;SETTING_3 .... meshfix_union_all_remove_holes = True**

Go to the "Mesh Fixes" settings and turn off "Remove all Holes".

I'll go ahead and close this as it is a setting issue.

closed 2024-10-27 by GregValiant

MichelBrai, 2024-10-27:
Bonjour,
Je suis désolé, je n'avais pas vu ça...

Merci beaucoup !

Meilleures salutations 

Michel Braito

MichelBrai, 2024-10-27:
Re-bonjour,

Je reviens vers vous, je suis désolé, cela ne fonctionne toujours pas.
J'ai d'abord essayé avec Cura en français, puis comme cela ne marchait pas  j'ai pensé que c'est un problème dû à la traduction et j'ai redémarré Cura en anglais.
Comme vous pourrez le voir sur l'image 1, que j'ai annexée à ce mail, les paramètres sont mis comme vous m'avez dit de faire.
Mais comme vous verrez sur l'image 2, les trous sont encore bouchés après le tranchage...

Merci encore pour votre aide et meilleures salutations
Michel Braito

GregValiant, 2024-10-27:
Github does not allow attachments via email.  You must attach directly to the site.  I did not get your last photos.

This is the file sliced with my regular settings.
[image]

This is slice with "Remove all Holes" turned on:
[image]

The model does not have any errors and should slice with the holes in it.
Load the model, set Cura up to slice, select "File | Save Project" command and a 3mf file will be created.
Zip that 3mf file and post the zip folder directly to this site (do not respond via email or I won't get the file).